LibreCAD/LibreCAD #1861: Add ability to change the size of grips

Questions2000, 2024-08-27:
Surprised this has not been mentioned, I apologize if I have overlooked this.

Maybe can add the ability to increase the size of the grips (squares that show up when items are selected).
The default size feels a little small.
Not sure if any other users agree?
Not sure if a setting for this would be considered to be added?

Thank You

sand1024, 2024-08-27:
well, yes, I think this is good idea to add an option that controls the size of reference point. 
I'll add it soon.

dxli, 2024-08-28:
One detailed technical thing, the grib size is chosen to be less than half of grid spacing, to make "free" and "grid" to work together smoothly.

Of course, point snapping could be different from entity snapping,  but some redesign needed with all snapping scenarios in mind.

sand1024, 2024-08-28:
oh, this is good point, thank you for your note!

Questions2000, 2024-09-01:
I am working on a post regarding updates to the Application Preferences windows.  This post will allow a space for this setting if decide to add it.
I will try to make that post as soon as possible.
I also updated my original post to include that I believe the proper name for these "grips" are called "nodes" which is what the polyline tools refer to them as.
Thank Thank

sand1024, 2024-09-02:
well, I'm sorry - yet I'd better suggest to delay you with this post for now. 
You see, preferences dialog is under development so far (with more logical rearranging and adding more settings), so I'm afraid that if you describe the current state of Preferences - it might be outdate. 

BTW - you've noticed that you were not aware of persistent dialogs. Usually, for all merge requests I submit, I put a short demo on my YouTube channel were I highlight at least most notable changes (as well as use corresponding note in MR). 
Something like this - https://youtu.be/dOW5O-37gjo - plus, more info regarding changes is usually there: 
https://github.com/LibreCAD/LibreCAD/tree/master/ChangeLogs

Questions2000, 2024-09-04:
Ahh that answers one of my questions, a while ago I was going to ask if anyone knew of a list of changes that occur with each move recent Continuous Build AppImage build.  Thank you for providing this.

Regarding the Preferences post I mentioned, no problem.
However, again something strange because as I said many ideas I had were literally things you added, and believe it or not, my post was going to be exactly what you just said, a rearranging and adding of more settings to preferences.

I actually already typed out a post for it, but it ended up being more than I thought anyone would be willing to consider so I did not post it, but now I am thinking maybe it could be helpful.
Obviously you do not have to add everything/anything I mention in it and obviously you guys know more about how you want to handle it but again maybe little things from my post could be helpful?
Let me think about it but I might make the post just because of the coincidence.
Thank you again

but there might be some things in it that could

sand1024, 2024-09-04:
well, just please write down any your ideas or suggestions - and I'm sure it will be possible to consider them and implement at least the most important/useful ones

Questions2000, 2024-09-21:
@sand1024
Just letting you know that I decided to make that settings post we were talking about.
The post was closed and moved to the following link in the Discussions section, if still interested.
https://github.com/LibreCAD/LibreCAD/discussions/1889
Thank you again for you help

Questions2000, 2024-10-18:
This was added, thank you.

closed 2024-10-18 by Questions2000