LibreCAD/LibreCAD #1864: Current Drawing Preferences - Dimensions - Width of Extension lines and Dimension lines, arrows & ticks is set to 0.00mm and cannot be changed

dellus, 2024-08-28:
Expected behavior: Width of Extension lines and Dimension lines, arrows & ticks should be able to be set as wanted

 Observed behavior: It is set to 0.00mm and cannot be changed in 2.2.1 rc and 2.2.2 alpha.  
 Even when opening a drawing made in 2.2.0, where it worked, it is specified and displayed as 0.00mm and cannot be changed.

Version: 2.2.1_rc3-16-g0a0a81de
Compiler: GNU GCC 9.4.0
Compiled on: Aug 23 2024
Qt Version: 5.15.2
Boost Version: 1.71.0
System: fossapup64 9.5

sand1024, 2024-08-28:
This is for "Current Drawing Preferences" dialog, correct? 

Could you please also check, whether the issue is only with setting width or other attributes are also affected? 

Also, could you please attach the file for which you see such behavior (it might be helpful, as settings for dimensions are not stored in the application settings - they are part of the file and is picked from variables stored within drawing file). 
Thank you.

dellus, 2024-08-28:
Testfile made in 2.2.0:
[dimlinwidth_220.zip](https://github.com/user-attachments/files/16788082/dimlinwidth_220.zip)
 It has not to do with the file. It seems the issue is how LC reads it.

Rendering and settings in 2.2.0:

[image]

Same file opened in 2.2.1:
[image]
You can alter the linewidth in the settings box but when you click OK the box closes and nothing has changed. If you open the settings again it's still 0.00mm.

I have not found other settings affected.

sand1024, 2024-08-28:
ok, thank you - I'll check and hope fix it soon

app4soft, 2024-08-30:
Same issue is present in LibreCAD `2.2.2_alpha`.

app4soft, 2024-08-31:
Additionally, in "Dimension" dialog its possible to set linetype for selected dimension, but it also has no effect on its appearance:

[image]

dxli, 2024-09-25:
@sand1024 

Do you think this fix should be in 2.2.1 as well?

The only 2.2.1 left on my list is the crashing on startup for windows  #1880

sand1024, 2024-09-25:
actually yet, it might be - I've found the reason for the issue (it seems its quite a long issue, according to the code - or probably I've just missed something and initially that functionality was correct ...) .

Anyway, while I'm not ready to make normal MR to the master at the moment (as the code is a bit in the middle of something),  I've pushed that fix in my repo, yet from the very first glance, I'm not sure it's simple to extract that fix from there.  

So probably it's simpler to backport that fix. 

The fix its quite straightforward:

1) void QG_DlgOptionsDrawing::setGraphic(RS_Graphic *g) {
there is a code like this

```
    RS2::LineWidth lineWidth = RS2::intToLineWidth(dimlwd);
    cbDimLwD->setWidth(lineWidth);
```

2) but - in void QG_DlgOptionsDrawing::validate() {

there is no back conversion, from LC line width to internal line width value (stored in drawing variable). 

Original code was like this: 

 ```
  graphic->addVariable("$DIMLWD", cbDimLwD->getWidth(), 70);
   graphic->addVariable("$DIMLWE", cbDimLwE->getWidth(), 70);
```
   
Yet correct one SHOULD perform conversion back (dimLwDLineWidth and dimLwELineWidth below may be inlined of course, I've used them for debugger mostly): 

```
 RS2::LineWidth dimLwDLineWidth = cbDimLwD->getWidth();
 int lineWidthDValue = RS2::lineWidthToInt(dimLwDLineWidth);
 graphic->addVariable("$DIMLWD", lineWidthDValue, 70);

 RS2::LineWidth dimLwELineWidth = cbDimLwE->getWidth();
 int lineWidthEValue = RS2::lineWidthToInt(dimLwELineWidth);
 graphic->addVariable("$DIMLWE", lineWidthEValue, 70);   
```

And back mapping is located in rs.cpp, like 

```
std::map<RS2::LineWidth, int> constructReversedMap(std::map<int, RS2::LineWidth> originalMap) {
    std::map<int, RS2::LineWidth>::iterator it;
    std::map<RS2::LineWidth, int> reverseMap;

    for(it = originalMap.begin(); it != originalMap.end(); it++)
        reverseMap[it->second] = it->first;

    return reverseMap;
}

const std::map<RS2::LineWidth, int> g_lineWidth2int = constructReversedMap(g_int2LineWidth);
```

and 

```
int RS2::lineWidthToInt(LineWidth lw){
    auto it = g_lineWidth2int.find(lw);
    return (it != g_lineWidth2int.cend()) ? it->second : -2;
}
```

Please refer to the following branch, that fix is also there: 
https://github.com/sand1024/LibreCAD/tree/ui_and_settings

sand1024, 2024-09-25:
oh, didn't noticed that reference to the commit above - I think changes I've described are there: 
https://github.com/sand1024/LibreCAD/commit/dba58f3172eeaadeea43f5c33e0c0dcaef6939c9

ksio89, 2024-09-29:
I'm not OP, but I confirm this has been fixed, thank you very much guys.

dellus, 2024-09-30:
It is not completely fixed, at least not for me in the linux appimage.
Out of the box the default setting still is 0.00mm. Yes it can be changed now. But when files made with an older LibreCAD version with a different setting are opened they again show 0.00mm.
And why at all the default now is 0.00mm? This will cause trouble for newbies and not so experienced users. In 2.2.0 the default was "By Block", which makes sense.

sand1024, 2024-10-01:
> But when files made with an older LibreCAD version with a different setting are opened they again show 0.00mm.

This is correct behavior - the linewidth for dimensions is stored directly in drawing, not in application settings.  So if file was created in later version - there will be 0.00, please update value, reopen drawing and check whether value was changed. 

>And why at all the default now is 0.00mm? This will cause trouble for newbies and not so experienced users. In 2.2.0 the default was "By Block", which makes sense.

Hm.... it's a good question. But why "By Block", not "By Layer"? 

I see that such change of default was intentional, yet I suppose the reason for it will be better addressed by @dxli..  I think it's part of commit 66b7058e

dellus, 2024-10-01:
- the linewidth for dimensions is stored directly in drawing, not in application settings. - Exactly. If I open a drawing made in 2.2.0 with a line width "By Layer" in 2.2.2 alpha1 173 then it shows 0.00mm. And in this case it also still cannot be changed!
It seems the data from the file is not read properly.

But why "By Block", not "By Layer"? - as I understand it the dimension as a whole is handled as a block, as it consists of several elements. Normally you want all the elements to have the same linewidth. But if you want the width of extension lines and dimension lines, arrows & ticks differently you can make it this way. Only the dimension text then stays as in the general definition.

sand1024, 2024-10-01:
>2.2.0 with a line width "By Layer" in 2.2.2 alpha1 173 then it shows 0.00mm. And in this case it also still cannot be changed!
It seems the data from the file is not read properly.

Oh, that's interesting... could you please share an example of such file? I feel that this is may be quite common issue, so probably some additional processing of old files is needed.

dellus, 2024-10-01:
A sample file is in the zip I have provided in my second post above: [dimlinwidth_220.zip](https://github.com/user-attachments/files/16788082/dimlinwidth_220.zip)

ksio89, 2024-10-16:
Sadly still happening on latest release candidate build.

app4soft, 2024-11-02:
- https://forum.librecad.org/2-2-1-finalization-tp5725538p5725541.html
  > > It's probably time to wrap up 2.2.1, since no new serious issues has been reported recently.
  > Issue 1864 is still not really fixed.
  > At least in the Linux appimage, despite claiming dxli to have pushed a commit which referenced this issue.
  > But #1922 is about rendering performance...

dellus, 2024-12-02:
I want to remind of this issue still not being fixed completely, with release 2.2.1 awaited.
In 2.2.1 rc4-4 still line width is fixed to 0.00mm and can not be changed. This refers to creating new drawings and old drawings made with 2.2.0.
In 2.2.2 alpha1 207 out of the box line width still is 0.00mm, but can be changed and saved now.
Old drawings from 2.2.0 with line width for instance 0.25mm open with 0.00mm also. They can be changed and saved here and then open properly in 2.2.1 rc4-4.
But funnily when opened in 2.2.0 are stated as 0.30mm!

I hope this bug will not be passed on into 2.2.1 final.

dellus, 2024-12-03:
Even if we accept there is no backward compatibility, my experience with 2.2.1 rc4-4 is that even in a new drawing the line width cannot be set to other than 0.00mm. I use the appimage in linux. Is it only a problem there? Maybe I get something wrong.
Could you test that appimage in a real linux and tell me it works right for you? I will give up then.

ksio89, 2024-12-10:
> Even if we accept there is no backward compatibility, my experience with 2.2.1 rc4-4 is that even in a new drawing the line width cannot be set to other than 0.00mm. I use the appimage in linux. Is [...]

No, this issue also present on Windows.

dellus, 2024-12-14:
@dxli 
thanks for backporting the fix in 2.2.1_rc4-6, 2.2.1 now is usable again.
The out of the box linewidth of 0.00mm now can be changed to a different numerical value, but still not to By Layer or By Block. And my question remains why the out of the box setting is not By Block as it has been previously and which makes sense.

dellus, 2024-12-19:
Sorry for having to report, but dimension line width now is 'Default', which is 0.00mm, not 'By Block', so nothing has changed.
2.2.1_rc5-2 and 2.2.2_alpha1-217.

dxli, 2024-12-20:
@dellus 

Could you elaborate a little bit? A test case dxf would help.

dxli, 2024-12-20:
@dellus 

sorry, I thought the issue was for "extension lines" only. Now, I see the issue is on "Dimension Lines".