LibreCAD/LibreCAD #1866: Why were commands renamed in build 2.2.1 rc3 and beyond?

ksio89, 2024-08-28:
I noticed that in build 2.2.1 release candidate 3 and beyond (like 2.2.2 alpha), several commands (and respective aliases) were renamed from what they were in 2.2.0.x continous build, for example:

``vertical`` => ``linehor``
``horizontal`` => ``linever``
``rectangle`` => ``linerec``

Was renaming long established commands really necessary, and will be the new names permanent? I know I that I can add aliases, but they won't be shown when cycling between commands with **Tab** key

Windows 11 Home 23H2
Version: 2.2.1_rc3-17-g40ce979f
Compiler: GNU GCC 12.2.0
Compiled on: Aug 26 2024
Qt Version: 5.15.2
Boost Version: 1.75.0

dxli, 2024-08-28:
vertical => linehor
horizontal => linever

Need to double check those two. 2.2.1 should be fixed for any issues like this.

ksio89, 2024-08-28:
Oops, I switched both commands, it's correct now. Anyways, several other command have been renamed, is there a good reason for this?

dxli, 2024-08-28:
We renamed a few due to reported issues and conflicts.

For example, offset is now moffset(modification) instead of line parallel, to be more conforming to cad users.

For power users, this might be a problem, so we should list all differences in the release document.

sand1024, 2024-08-29:
well, as for commands - it might be indeed that some commands were renamed due conflicts.  

I think the overall visibility of commands for the user also requires may be improved. That might be especially important for new users. 

Of course, so far it's possible to check the list of commands via TAB key, but the information provided there is minimalistic. 

Probably we need some UI that will displays available commands to the user, with an additional help/attributes (considering that some of them may be overridden via alias). 

So in general, internally it will be necessary to define more expanded meta-information that describes commands and add UI for it. 

So far it's just overall ideas, yet I've noted this.