LibreCAD/LibreCAD #1867: Setting linear dimensions "DIMSTYLE"

app4soft, 2024-08-30:
### Expected behavior

LibreCAD should support all standard DXF dimensions' `DIMSTYLE` variants, and there should be an option to switch `DIMSTYLE` for selected dimensions.

### Observed behavior

LibreCAD has no option to change `DIMSTYLE` for linear dimensions.

### Steps to reproduce or sample file

1. Create drawing and add linear dimension;
2. Try change its `DIMSTYLE`.

### Operating System and LibreCAD version info

All versions and platforms.

sand1024, 2024-08-30:
yep, that's a good point. For sure, dimensions support should be extended, and, for example, support of ordinate dimensions in addition to linear also needed. 

However, support of all dimension types is not a quick feature to implement so it may require some time. 

Anyway, thank you for noting this issue - and it's great that you've also added references there, it will be simpler to implement support. 

I think we definitely need to support additional dimensions types.

app4soft, 2024-08-30:
> For sure, dimensions support should be extended, and, for example, support of ordinate dimensions in addition to linear also needed.

I just updated issue description and added AutoCAD's "DIMENSION: Modify" dialog screenshot.

LibreCAD already has some of `DIMSTYLE` variables in "Current Drawing Prefernces > Dimensions" (applied to all dimensions in drawing), so adding missing options should be not too hard (as LibreCAD can render some `DIMENSIONS` features, where there is no option to set it manually, for example, invert dimension arrows direction).

But there is very limited set of options in "Dimensions" dialog (if use "Prefernces" on selected dimensions), so customizing `DIMSTYLE` per each dimension in LibreCAD is almost impossible as for now. I suppose there should be more options (already presented in "Current Drawing Prefernces > Dimensions") added into "Dimensions" dialog for selected dimension customizing.

closed 2025-04-27 by u2fly