LibreCAD/LibreCAD #1868: Impossible to remove dimension label prefix once entered

app4soft, 2024-08-30:
### Expected behavior

Adding and removing label prefix/suffix should work as expected.

### Observed behavior

Actually if for linear dimension add some custom prefix/suffix, and then try to remove such prefix it still persists and always shown in "Dimension" dialog "Label" field.

### Steps to reproduce or sample file
1. Create/Open dawing;
2. Add new linear dimension;
3. Select it and open "Dimension" dialog;
4. In "Dimension" dialog add any letter as prefix/suffix in "Label" field, and press <kbd>OK</kbd>;
5. Re-open "Dimension" dialog, remove "Label" all custom prefix/suffix, and press <kbd>OK</kbd>.

### Operating System and LibreCAD version info

- Version: 2.2.2_alpha1-150-gdc895a29
- Compiler: GNU GCC 9.4.0
- Compiled on: Aug 30 2024
- Qt Version: 6.6.3
- Boost Version: 1.71.0
- System: Debian GNU/Linux 12 (bookworm)

dxli, 2024-08-31:
I cannot reproduce this one.

app4soft, 2024-08-31:
> I cannot reproduce this one.

More details: there is special case if type in `R` letter as a prefix for linear dimension.

Steps to reproduce:

1. Add dimension;
2. Edit dimension label by adding any word as prefix (e.g., `Any` or other) and click <kbd>OK</kbd>;
3. Re-edit dimension label by removing all prefixes and click <kbd>OK</kbd>;
4. Re-edit dimension label by adding `R` as prefix and click <kbd>OK</kbd>;
5. Try to remove prefix `R` or replace it with other prefix.

Screencast:
[simplescreenrecorder-2024-08-31_16.19.45.webm](https://github.com/user-attachments/assets/8aa783bf-f61f-4a09-b461-d5fee8314fdc)

closed 2024-08-31 by app4soft