LibreCAD/LibreCAD #1869: Calculator wrongly calculates trigonometric functions

app4soft, 2024-08-31:
### Expected behavior

Calculator (`cal` mode in Cmd widget) should work as expected.

### Observed behavior

Calculator wrongly calculates some of trigonometric functions.

### Steps to reproduce or sample file

1. Turn ON "Calculator" mode in Cmd widget;
2. Calculate `sin(90)`, `cos(90)`.

LibreCAD screenshot:
[image]

Trigonometric functions (correct values):
[image]

### Operating System and LibreCAD version info

- Version: 2.2.2_alpha1-151-g614f82dc
- Compiler: GNU GCC 9.4.0
- Compiled on: Aug 31 2024
- Qt Version: 6.6.3
- Boost Version: 1.71.0
- System: Debian GNU/Linux 12 (bookworm)

sand1024, 2024-08-31:
well, need to check this in more details, yet I suspect that parser expects angles in radians, and you in your example angles in degrees are given.. 
sin(90 rad) = 0.893996664

Basically, calculator - seems to be the whole area to improve...

app4soft, 2024-08-31:
> `sin(90 rad) = 0.893996664`

Its impossible to input `sin(90 rad)` (because it has spaces, <kbd>Space</kbd> key recognized same as <kbd>Enter</kbd>).

I could input `sin(90deg)` (without spaces, it then converts it to `sin(90*pi/180)`) and it calculates ok.

But for `cos(90deg)` (aka`sin(90*pi/180)`) it still calculates wrongly.

Screenshot:
[image]

dxli, 2024-08-31:
@sand1024  we may consider switching from MuParser to MuParserX

I remember I disabled "space" as enter in the command widget.

The screenshot shows MuParser works as intended. The calculator is not intended to do floating points instead of exact math.

sand1024, 2024-08-31:
@dxli oh, I'll take a look to MuParserX. 

I feel that the calc may be extended in general - say, have better support of arithmetic's and function (which may be domain-specific).  
Off course, it hardly will be as powerful as, say, one in AutoCAD, yet some nice features there may be handy. 
In addition to support of more functions, for example, I can imagine some kind of auto-suggest for commands (like autocomplete in IDEs).  
So basically I think this is an area to think about and revise.

app4soft, 2024-08-31:
> I remember I disabled "space" as enter in the command widget.

In latest LibreCAD `2.2.2_alpha` (AppImage) <kbd>Space</kbd> still acts like <kbd>Enter</kbd>.

dxli, 2024-08-31:
@app4soft 

Options > Application Preferences > Defaults > Keyboard Settings > Evaluate Command when SPACE BAR is pressed

app4soft, 2024-08-31:
> Options > Application Preferences > Defaults > Keyboard Settings > Evaluate Command when SPACE BAR is pressed

Just tried to toggle it ON and OFF — has no effects, <kbd>Space</kbd> still acts like <kbd>Enter</kbd>.

dxli, 2024-08-31:
@app4soft ,
 as preferences in "Application Preferences", need to restart the app to take effects  after changes made.

app4soft, 2024-08-31:
> > Options > Application Preferences > Defaults > Keyboard Settings > Evaluate Command when SPACE BAR is pressed
>
> Just tried to toggle it ON and OFF — has no effects, <kbd>Space</kbd> still acts like <kbd>Enter</kbd>.

UPDATE: To clarify, if enter Cmd line with <kbd>Ctrl</kbd>+<kbd>m</kbd> shortcut, <kbd>Space</kbd> inputs spaces in Cmd, and do not act like <kbd>Enter</kbd> key.

But if start Cmd with <kbd>@</kbd> shortcut or by click with mouse cursor in Cmd input field, <kbd>Space</kbd> still acts like <kbd>Enter</kbd>, even after restart.

Might be related to this: https://github.com/LibreCAD/LibreCAD/issues/1115#issuecomment-1790810569

dxli, 2024-08-31:
@app4soft just tested with both 2.2.2 and 2.2.1, it works for me, and actually WITHOUT the need to restart the app.

1. start LC;
2. draw line by points;
3. activate the command line, try to type in coordinates like, "100, 100".

If the spacebar triggers evaluation, the coordinate input failed right after the space bar is pressed.

If spacebar triggering evaluation is disabled, the coordinates, can be typed with a space after the comma, and no error is triggered.

app4soft, 2024-08-31:
> 1. start LC;
> 2. draw line by points;
> 3. activate the command line, try to type in coordinates like, "100, 100".

I can't input `100, 100` in Command input field — Cmd outputs me next message after pressing <kbd>Space</kbd>:
```
100,
Expression Syntax Error
```

> If the spacebar triggers evaluation, the coordinate input failed right after the space bar is pressed.
>
> If spacebar triggering evaluation is disabled, the coordinates, can be typed with a space after the comma, and no error is triggered.

Does not work for me. Tried both options — issue is the same, and just can't input spaces in Command.

app4soft, 2024-08-31:
UPDATE: 

0. Just replaced `LibreCAD-2.2.2_alpha1-155-g62f875c8-x86_64.AppImage` with `LibreCAD-2.2.2_alpha1-156-g81421d53.AppImage`;
1. Started LibreCAD;
2. Draw line by points;
3. Activated the command line and typed in coordinates `100, 100`.

All works now, without warnings in Cmd (spacebar triggering evaluation is disabled).

I'm not sure what was wrong with previous build.

closed 2025-04-18 by u2fly