LibreCAD/LibreCAD #1870: Extended axis lines dissapears if origin is out of canvas

app4soft, 2024-08-31:
### Expected behavior

Extended axis lines should be shown, even origin is moved out of actual window canvas.

### Observed behavior

Grid "Extension lines" dissapears when origin is moved out of actual window canvas.

### Steps to reproduce or sample file

1. Enable "Extend exis lines" in "Current drawing > Grid";
2. Pan canvas to move origin out of actual window canvas.

Screencast:
[simplescreenrecorder-2024-08-31_14.18.41.webm](https://github.com/user-attachments/assets/c5673ba4-ba14-41b9-9e24-6ccf9c027f8a)

### Operating System and LibreCAD version info

- Version: 2.2.2_alpha1-151-g614f82dc
- Compiler: GNU GCC 9.4.0
- Compiled on: Aug 31 2024
- Qt Version: 6.6.3
- Boost Version: 1.71.0
- System: Debian GNU/Linux 12 (bookworm)

sand1024, 2024-08-31:
oh, what a nighty bug... thank you for reporting, it's weird that I've relied on merge and missed it. 
Will be fixed in next build.

sand1024, 2024-10-26:
@app4soft  - please check and close if it works fine now

closed 2024-10-27 by app4soft