LibreCAD/LibreCAD #1872: Action Shortcut Mappings starts with a horizontal scrollbar

Questions2000, 2024-08-31:
LibreCAD-2.2.2_alpha1-150-gdc895a29.AppImage
Linux / Debian 12 based distro / GTK based DE
QT5 5.15.8

First off, WTF! who created this, wow!
I had an idea for something similar that I was debating on mentioning because I didn't think anyone would work on it but this is way more advanced than my idea, what a great job, thank you.

The only minor thing I noticed, could be a system setting or something that not everyone experiences? but I noticed that the window size opens up with a width that just slightly causes the horizontal scroll bar to show up.  If I adjust the width of the window slightly the horizontal scroll bar goes away.  Was wondering if the windows width can be adjusted to automatically open just slightly wider to prevent this.

Again great job, thank you for putting so much work into this.

app4soft, 2024-08-31:
> but I noticed that the window size opens up with a width that just slightly causes the horizontal scroll bar to show up

Guess, it could be fixed with small change in `lc_actionsshortcutsdialog.ui`:

- https://github.com/LibreCAD/LibreCAD/blob/master/librecad/src/ui/dialogs/settings/shortcuts/lc_actionsshortcutsdialog.ui

I'm actually working on minor UI cleanup in `*.ui` files (https://github.com/LibreCAD/LibreCAD/pull/1855), but not yet touched shortcuts dialog, as it is a fully new dialog window UI layout added by @sand1024.

[image]

sand1024, 2024-08-31:
well, thank you.
BTW - as far as I remember, that dialog is resizable, so could you try to resize it and check how that input behaves? 

Plus, if you enable settings for making dialogs persistent (in Application Preferences) - the size and position of the dialog (as well as other ones) will be stored and so on the next invocation the dialog should be displayed in specified size and position.  

I've noticed that fighting with dialogs layouts for various resolutions, fonts, and operating systems may be quite challenging, so decided to add persistent dialogs positions. With that feature, each user may adjust dialogs for its own system and preferences.

app4soft, 2024-08-31:
@sand1024 I suppose, it could be set to autoajust to `tvMappingsTree` by setting right `QWidget > sizePolicy > Horizontal Policy`, but I'm not sure how to apply it corretly in this case.

Questions2000, 2024-09-01:
@app4soft
Thank you for your responses

@sand1024
Ummm, okay double wow!
I was not aware of this setting, I tried it out and it works great, great idea to add this to solve the possible dialog based issues that you mentioned.  I like this even better then what I requested because in this case it remembers what size I want personally.
Again I cannot say enough good things about this shortcut mapping addition, has solved so many things I was looking to achieve already and I have only just started using it, really great.

Not sure what to do I guess I will close this post?, I think you guys probably have the ability to re-open issues if there is anything here that is desired to be updated.  If you guys cannot re-open this just let me know if you need me to.

Before I go, while on the subject, I had mentioned this in another post, but I will mention it again, and probably keep on mentioning :) 
I had a list of many items of things to fix / update / suggestions and all of a sudden after updating to the lastest AppImage I noticed that most of them have been corrected/added.
Turns out you were a big part of many of these items, so thank you so much for these as well, so many great updates thank you for working on all of this, very helpful and greatly appreciated :)

closed 2024-09-01 by Questions2000

sand1024, 2024-09-02:
thank you for your feedback and appreciation! Well, I hope with time LibreCAD will be a bit more convenient. 

Well, I think there is something still left in your list - if it is so, please let me know your ideas.

Questions2000, 2024-09-04:
Oh no problem thank you.
Well, honestly I have a good amount of ideas, so I do not think you know what you are asking for :)
Not to say that they are all good either :)
I have already been making a lot of posts so I am trying to hold back overloading this site / developers.
I just do not want to upset anybody.
But like I said thank you again, really amazing work.

dxli, 2024-09-04:
Yes, @sand1024 has been adding lots of quality changes.

Hope we can get 2.2.2 a new height.

I want to concentrate myself on algebra and geometry part:

1, hyperbola, a ten year late feature for this project;
2, hatch and possible topology feature;
3, new curve types based on splinepoints, such as spirals, conic section offset (beyond circles), or direct plot by equations.

sand1024, 2024-09-05:
@dxli 
well, thinking about geometry - so far I have just a general idea to consider and think about. 

I think it might be really nice feature if we'll add the user possibility to do isometric drawing.  
Currently, even for isometric grid enabled, all coordinates are still cartesian, and grid is just used for simplification of snapping along isometric coordinates. However, the coordinates of entities itself are still cartesian and isometric grid is rather visual guide and snapping tool.  So as for now - it's rather pseudo-isometry support.  

From the other hand - as far as I can see, it should be possible to translate coordinate system - so potentially, it should be possible to create entities in cartesian coordinates, but render them in isometric. 

Thus it will be simpler to create left/right etc. projections of parts, plus, if one day support of viewports will be extended - it may bring additional benefits there. 

I suspect that addition of coordinate system translation may be done in transparent way (as anyway, mapping of drawing/UI coordinates is encapsulated within GraphicView, so the overall logic of actions and generic math will still operate with cartesian coordinates. However, most probably the renderer (RS_Painter) may require an additional support for rendering complex entities  

In general, translation of coordinates for primitives like line/point is straightforward, transforming circle/arc to ellipse/elliptic arc is also transparent, yet for more complex things like spline, ellipse, hyperbola - I suppose it's the area where additional math support will be really useful. 

Please consider this.

dxli, 2024-09-05:
It's actually trivial to support entering 3d coordinates in isometric and place them. 

The trouble is the geometry engine assumes everything in the z=0 plane.

What's the best way to handle this? Internal 2d, and input display at 3d, with a user selected bijection mapping (say, drawing on the yz-plane view).

app4soft, 2024-09-05:
> probably there some other tools that supports such a mode.

In _SolveSpace_ (since `3.1` release) there is the _"Show Exploded View"_ mode (shortcut: <kbd>\</kbd>) for 2 sketches, and its possible to set Z-offset between each entities in _"Property Browser > view > explode distance"_ (default value is `5.00` mm)

- https://www.github.com/solvespace/solvespace/pull/1003

Screenshots before and after:

[image]
[image]
[image]

sand1024, 2024-09-05:
@app4soft 
Yes, I've mean something like this
@dxli 
that's funny, just found that ZCad is not dead - you can check the latest version there https://github.com/zamtmn/zcad/releases
It's written in Pascal, so I'm not sure sources that may be helpful for some inspiration, but as for the overall set of features - why not. There such mode as I've described is available via menu View->View->View .. isometric.

app4soft, 2024-09-05:
Discussion here is now not related to actual issue at all, so here is an Isometric projection tool feature request:

- https://github.com/LibreCAD/LibreCAD/issues/1882

And here are other Isometric-related issues:

- https://github.com/LibreCAD/LibreCAD/issues?q=is%3Aissue+is%3Aopen+isometric