LibreCAD/LibreCAD #1874: Cmd input field cleared by fast move of mouse cursor from right to left

app4soft, 2024-08-31:
### Expected behavior

Moving mouse cursor above Cmd widget input field should not clear typed command.

### Observed behavior

Moving fast mouse cursor from right to left above Cmd widget input field clears typed command.

Screencast:
[simplescreenrecorder-2024-08-31_22.57.10.webm](https://github.com/user-attachments/assets/dd56b769-f33e-4fe4-b2c8-890caa581e6c)

### Steps to reproduce or sample file

1. Input command in Cmd widget input field (and do not press <kbd>Enter</kbd>);
2. Move fast mouse cursor above Cmd widget input field from right to left.

### Operating System and LibreCAD version info

- Version: 2.2.2_alpha1-155-g62f875c8
- Compiler: GNU GCC 9.4.0
- Compiled on: Aug 31 2024
- Qt Version: 6.6.3
- Boost Version: 1.71.0
- System: Debian GNU/Linux 12 (bookworm)

sand1024, 2024-08-31:
hm.. most probably with fast move the cursor enters graphic view (central widget). At least, as far as I can see, fast moving closer to the right corner of the input does not clear the text in the command line. 

Anyway, it's a good question why the command line is cleared when graphic view receives focus...  I'll take a look. 

Yet in general, I've considering revision of focus management together with the overall keyboard support.

dxli, 2024-09-01:
@sand1024 

Could you review my commit? Something really weird there, and I have no clue whether this commit would break something important.

sand1024, 2024-09-19:
hm... yes, I've checked your commit - but as far as I can see, setCommand on command widget just updates prompt label above the input field in Cmd. 

Thus updateMouseWidget() in DialogFactory (despite it's name) actually updates command prompt message for two widgets - in MouseWidget and for Cmd. 

This is valid behavior and very old functionality, I suppose. So I'd better uncomment this piece. 

As for the issue - I suspect it's related to mouse enter event. 

Yet what it's weird enough - I've tried to reproduce it on my last build and with no luck... so need  more investigation of this

u2fly, 2025-04-18:
In latest 2.2.2.x build its fixed.