LibreCAD/LibreCAD #1875: Allow Lines grid for Isometric grid and fix its look

app4soft, 2024-09-01:
### Expected behavior

Isometric Grid should be available in both types, Points and Lines.

### Observed behavior

Isometric Grid only available in both Points type.

But there is some bug which allows to switch to Isometrci Grid by different steps, and resulted Isometric Grid looks messy.

### Steps to reproduce or sample file

1. Open "Current Drawing Preferences";
2. Change "Grid type" to "Lines" (**option to switch to "Isometric Grid" is disabled now**);
3. Change "Grid type" back to "Points";
4. Switch to "Isometric Grid";
5. Change "Grid type" to "Lines".

Screenshot:
[image]

### Operating System and LibreCAD version info

- Version: 2.2.2_alpha1-156-g81421d53
- Compiler: GNU GCC 9.4.0
- Compiled on: Aug 31 2024
- Qt Version: 6.6.3
- Boost Version: 1.71.0
- System: Debian GNU/Linux 12 (bookworm)

sand1024, 2024-09-01:
yes, so far lines grid support only orthogonal grid - that comes from original commit ed31f50856593b7e32e8f810b0073b985d636f76

Actually, it's quite a quick implementation for lines grid in that commit that supports only orthogonal grid. 

For sure, support of lines grid drawing should be moved to general grid drawing logic, so yes, that will be more consistent as we correct this. 

Yet as for double drawing of both grids - that's weird behavior indeed, thank you for reporting, that will be fixed in next commit.

app4soft, 2024-09-01:
> yes, so far lines grid support only orthogonal grid - that comes from original commit https://github.com/LibreCAD/LibreCAD/commit/ed31f50856593b7e32e8f810b0073b985d636f76

Are all the features from https://github.com/LibreCAD/LibreCAD/pull/1831 already merged into `master`?

Also, there would be good to has few extra options:

- Show extended lines only for positive values, i.e. Up and Right (it might be also related to https://github.com/LibreCAD/LibreCAD/issues/1870);
- Show origin as <sup>`X`</sup>`L`<sub>`Y`</sub>, i.e.:
  > X
  > ↑
  > ˈ-→ Y
- Show Metagrid as Lines and Grid is Points (similar to QCAD), i.e.:
  [image]

sand1024, 2024-09-01:
> Are all the features from https://github.com/LibreCAD/LibreCAD/pull/1831 already merged into master?

this one?
https://github.com/LibreCAD/LibreCAD/pull/1497

Yes, as far as I know...  As I've mentioned, it was quite a fix support of lines grid...

app4soft, 2024-09-01:
> Yes, as far as I know... 

Why `#1831` pull still open, if all of the features from `#1497` pull already merged into `master`?

Also, as just added into previous comment (https://github.com/LibreCAD/LibreCAD/issues/1875#issuecomment-2323278251), I can't find a way to set show Preview Reference Point cross with a hollow squre at center.

Screenshot from `#1497` pull description:
  [image]

sand1024, 2024-09-01:
> Show extended lines only for positive values, i.e. Up and Right (it might be also related to https://github.com/LibreCAD/LibreCAD/issues/1870);

Hm... actually don't really get it.  You mean that X axis should be draw say, for values from (0.0 to 0..infinity and same for Y axis)?
 
Technically - it's easy to add, but not sure I understand the the benefit.  Yet if it's convenient for somebody - sure, can add an option, thought

>1.1. Show grid origin as XLY;

Hm... is it really necessary? basically, it's convenient feature for different projections, but without full-fledge support of isometric drawing mode (with translations of entered and rendered coordinates to proper projection) it might be just a graphical embellishment without some real practical sense, I'm afraid.

sand1024, 2024-09-01:
>Why #1831 pull still open, if all of the features from #1497 pull already merged into master?

I don't know. 

@dxli  - could you please close it? 

>I can't find a way to set show Preview Reference Point cross with a hollow squre at center.

well, not sure I understand this properly.  Here you refer to the shape of cursor or to points that are drawn as reference in preview of operations?  For reference points, all shapes of points supported may be selected in Application Preferences. Except a couple of not visible types of points, all shapes of points that may be drawn within drawing (for rendering point) are supported as reference points. 

Yet if you refer to the shape of the cursor - no, there is no such option. In AutoCad, the square in the middle has special meaning - it also shows the are where snapping algorithms will work (say, snapping to entities, endpoints etc).  LibreCAD does not support this.

app4soft, 2024-09-01:
> > I can't find a way to set show Preview Reference Point cross with a hollow squre at center.
>
> well, not sure I understand this properly. Here you refer to the shape of cursor or to points that are drawn as reference in preview of operations?

Here are settings I used to set Preview Selectable Refernce point look like in ACad: https://github.com/LibreCAD/LibreCAD/pull/1497#issuecomment-2323093511

But combining "Preview Selectable Refernce" cross together with "Current Drawing Preferences > Point" hollow square, results in crossed square, not a hollow square with extension lines.

[image]

app4soft, 2024-09-01:
> You mean that X axis should be draw say, for values from (0.0 to 0..infinity and same for Y axis)? Technically - it's easy to add, but not sure I understand the the benefit. Yet if it's convenient for somebody - sure, can add an option, thought.
>
> >  Show grid origin as <sup><sup>`Y`</sup></sup>`L`<sub><sub>`X`</sub></sub>
> Hm... is it really necessary?

I think, it a part of what is requested in https://github.com/LibreCAD/LibreCAD/issues/1477 feature request is created for — full customization of UI (with possibility to mimic to other CADs, so it could be usefull for user switching from other CADs to LibreCAD).

Axis origins in other CADs:
[image]

sand1024, 2024-09-01:
> > > I can't find a way to set show Preview Reference Point cross with a hollow squre at center.
> > 
> > 
> > well, not sure I understand this properly. Here you refer to the shape of cursor or to points that are drawn as reference in preview of operations?
> 
> Here are settings I used to set Preview Selectable Reference Point look like in ACad: [#1497 (comment)](https://github.com/LibreCAD/LibreCAD/pull/1497#issuecomment-2323093511)
> 
> But combining "[Application Preferences > Preview Selectable Refernce Points](https://github.com/LibreCAD/LibreCAD/pull/1497#issuecomment-2323093511)" _cross_ together with "Current Drawing Preferences > Point" _hollow square_ results in crossed square, not a hollow square with extension lines.
> 

Still not sure I understand this. 

1) Current Drawing Preference > Point - defines how normal point entity will be drawn on drawing rendering; 
2) Application Preferences > Preview Selectable Reference Points - defines how to draw temporary refence points that occurs during specific actions execution and allows to understand expected action's results better. Such temporary points are something like endpoints, center of arc, intersection points, source/target points etc. - that depends on specific action and entity to be created. 

So combining them will not bring any effect - they are just different settings for different purposes. 

Also, both of the do not (and, in general, should not) affect the shape of mouse cursor.

sand1024, 2024-09-01:
> I think, it is part of what is requested in #1477 feature request is created for — full customization of UI (with possibility to mimic to other CADs, so it could be usefull for user switching from other CADs to LibreCAD).
> 
> Axis origins in other CADs:

Yes, for sure drawing such oriented axis is helpful  if coordinate system may be changed - so the user should know current X and Y direction. 

However, in LibreCAD - the coordinate system is fixed, so it's not possible, say, direct Y from top to bottom.  

That's why I suppose that drawing axis in such way may be also misleading...  

Of course, if more flexible coordinate system will be supported (isometric, or axis-mirroring) - such way of drawing Zero is more then needed. For now... well, if I'll have some spare time, I could add it (plus, appropriate preferences are needed to control at least shape and size..), but for now I can't say it will have high priority.

app4soft, 2024-09-01:
> well, if I'll have some spare time, I could add it (plus, appropriate preferences are needed to control at least shape and size..), but for now I can't say it will have high priority.

Of course, it is optional feature not on hight priority, but would be nice addition to LibreCAD.

BTW, Actual _Lines_ type issue for "Isometric Grid" is a major issue (as it really a bug).

Also, an option to set _Lines_ type for "Meta Grid", while keep "Grid" as _Points_, is really wanted feature.

sand1024, 2024-09-01:
> BTW, Actual Lines type issue for "Isometric Grid" is a major issue (as it is really a bug).

Yep, I'm working on it. 

> Also, an option to set Lines type for "Meta Grid", while keep "Grid" as Points, is really wanted feature.

Well, actually the MetaGrid is already drawn using line. The thing is that the style for that line is hardcoded and is dotted line. 

I think I can simply add the options to settings that will let the user to control line type and linewidth used for drawing metagrid lines.

app4soft, 2024-09-01:
> I think I can simply add the options to settings that will let the user to control line type and linewidth used for drawing metagrid lines.

Yes, it would be good addition.

sand1024, 2024-10-26:
@app4soft  - I suppose we can close it now?

closed 2024-10-27 by app4soft