LibreCAD/LibreCAD #1877: Polyline append node tool creates a new polyline on top of existing

Questions2000, 2024-09-01:
LibreCAD-2.2.2_alpha1-150-gdc895a29.AppImage
Linux / Debian 12 based distro / GTK based DE
QT5 5.15.8

I noticed when testing out the polyline tools that the "Append Node" which I believe is supposed to just add to an existing polyline, ends up creating a new polyline copy on top of the existing polyline it was created from.
After running the command, escape out and then click on the new polyline and drag it off to the side.  You will noticed the existing polyline is under it.

Thank You

sand1024, 2024-09-02:
Oh, that's good point, thank you  - indeed, this is a bug that I'll fix with next MR

Questions2000, 2024-10-21:
This appears to be fixed now, thank you for this update, I will close

closed 2024-10-21 by Questions2000