LibreCAD/LibreCAD #1878: Add more options to angle constraint setting

Questions2000, 2024-09-01:
Amazing that I had many items I was going to suggest to be added to this program and now all of a sudden many of them are included, a setting for angle constraints was one of them.

I noticed in the following location you can now set Angle Constraint settings
Application Preferences - Defaults - Angle snap single step

Sometimes users may not want such small degree variations for various reasons.
One reason especially is when drawing short lines or moving items a short distance, the shorter the line the more jagged that action can be because more angle snaps are occurring.

`CURRENT DEGREES VALUES OFFERED`
1
3
5
15

`NEW DEGREE VALUES THAT COULD BE ADDED`
30
45
90 = allows this constraint to only function horizontally & vertically

Also not sure if needed but maybe in the setting can add a check box to this option to be able to turn it on/off.  Again might not be needed but could maybe offer some benefits such as....
-Obviously allowing user to decide to turn this off
-Not sure if any connection but now with ability to set custom program hotkeys users may use Shift for other actions
-This would align this setting item with all the other check box items in the list where it is located
(Similar to an item I mentioned here LINK)

Hopefully others may find these new options useful
Thank You

  (I also noticed that this constraint tool was added to other tools such as the Stretch command, thank you)

I also noticed that this constraint tool was added to other tools such as the Stretch command, thank you)

sand1024, 2024-09-02:
well, adding new options is not an issue, yet
> 1) 90 = allows this constraint to only function horizontally & vertically 
this may be achieved with snap constraints restrictions, just use restrict both in snap toolbar. No problem to add them, however
> 30, 45
Ok, will add them. Actually, these values are not included initially since such angle may be achieved using smaller angles (like 30=15+15). 

>> Also not sure if needed but maybe in the setting can add a check box to this option to be able to turn it on/off. Again might not be needed but could maybe offer some benefits such as....

Hm... here I'm really not sure I understand why such checkbox will be reasonable - angle snap is usually invoked as the user pressed key modifier (mostly SHIFT) together with mouse move/click. 

so if shift is not pressed - no angle snap at all. 

Basically, such checkbox may be potentially useful if we'll support angle restriction (similar to horizontal/vertical restriction) - but so far angle restriction is not supported.

Questions2000, 2024-09-02:
@sand1024
Thank you for your reply.
Some of the formatting of your post is not coming through on my side so I apologize if my following response does not make sense.

`30 degrees`
I agree, I just randomly threw that into this request without thinking, was just completing the sequence of numbers in the list all the way up to 90 degrees, I just felt bad that 30 degrees was the only number in the sequence not good enough to make it on the list :)
The only benefit to this would be if user had a specific desire/need to draw/move items by 30 degrees for a specific task.
Would not be something most users would have set/on by default.
 
`45 degrees`
This one I think would be most important if willing to add this to the list.
This allows to not have a ton of angle snaps occuring with short lines/movements, but allows for horizontal, vertical & 45 degree angles.

`90 degrees`
I agree with what you are saying about ortho.
However the benefit could be that this would be able to be engaged with a simple modifier key hit with your left hand, holding down the shift key to temporarily activate it "quickly" and then releases/turns off right away.
Also many graphical programs use Shift key in this way as some programs do not offer any extra angles to snap to.
This one is up to you if feel it is necessary to have on the list.  I think as long as "45 degrees" could be added, that would be fine with me.

Check Box
I know, but then all the setting will not be aligned in the list.... :)
I agree no need no for it :)

Thank you again for all your help

Questions2000, 2024-09-10:
Having all these angles would just cover all the possible needs for a feature like this.
Granted not all are needed for general/typical use but maybe for specific needs.
Not always about the fact that smaller degree values can be added up to also cover larger degree values, in certain use cases user may just want to be able to restrain to specific angles only.

So it is hard to just pick only some values and not others.
I mean 1, 3 & 5 degrees are included currently so hard to say that others should or should not be included.  Does anyone use 1, 3 & 5 regularly?

Anyways like I said the main one I would be looking for is 45 degrees if possible.
The rest you guys will have to decide upon.

Thank you again

Questions2000, 2024-10-18:
This was added, thank you.

closed 2024-10-18 by Questions2000