LibreCAD/LibreCAD #1880: LibreCAD does not open, only shows splash screen

ksio89, 2024-09-03:
### Expected behavior
application opens

### Observed behavior
I'm experiencing the same issues as #1817 on all 2.2.1 release candidate and 2.2.2 alpha builds: the Loading... splash screen is shown but the application does not open. Uninstalling and reinstalling the same or newer builds does not fix it. 

The only workaround I've found is uninstall LC, install **2.2.0.2-14-g10ad9444** and then reinstall newer builds, but eventually it will stop launching again.

### Steps to reproduce or sample file
start the application

### Operating System and LibreCAD version info
OS: Windows 11 Home 23H2
LibreCAD version: 2.2.1_rc3-20-gd7446c30-win64

app4soft, 2024-09-03:
> OS: Windows 11

Might be issue of LibreCAD's Windows builds or Windows-specific issue.

On Linux all works.

dxli, 2024-09-03:
@ksio89 

Could you run from  cmd and with the option -d6 ?

Please upload the console output here.

I cannot reproduce this issue here.

ksio89, 2024-09-04:
I fixed the issue by uninstalling LibreCAD and then cleaning all remaining data and registry entries with Revo Uninstaller. It was some configuration created on 2.2.0 that was conflicting with 2.2.1 release candidate and 2.2.2 alpha builds. 

I'll keep using 2.2.1 rc and and backup LC folder on AppData, I just hope it won't happen again when it gets a final release.

dxli, 2024-09-04:
Let's make this one as a blocker for the 2.2.1 release.

We process the following at startup:

Command alias；
Block Library;
Fonts;
General preferences for the app (not for drawings).

Let's review each, and at least generate informative error messages upon crashes.

Questions2000, 2024-09-04:
@ksio89 
Not sure if this is helpful / the same issue you are experiencing but also take a look at this #1809 post.
Not sure if you edit your librecad.alias file to setup custom hotkeys?  but it could have been what your Revo Uninstaller took care of.  It might have deleted your older librecad.alias which maybe was the cause?
There appeared to be a bug regarding the librecad.alias file with newer releases.
Hopefully this is helpful, if not I apologize if this has nothing to do with your issue.

dxli, 2024-09-04:
Alias crash should have been fixed, 2.2.1 or 2.2.2

If you have an alias file causing the crash, please share.

ksio89, 2024-09-05:
> @ksio89 Not sure if this is helpful / the same issue you are experiencing but also take a look at this #1809 post. (I had made a link for this in my 1817 post, just mentioning just in case it was not seen) Not sure if you edit your librecad.alias file to setup custom hotkeys? but it could have been what your Revo Uninstaller took care of. It might have deleted your older librecad.alias which maybe was the cause? There appeared to be a bug regarding the librecad.alias file with newer releases that has now been corrected. (Using Linux so not sure if Windows version experienced the same issue)
> 
> Hopefully this is helpful, if not I apologize if this has nothing to do with your issue.

I also suspected that it was a custom libreacad.alias that was preventing LC from opening, but the crash persisted even with default alias file.

dxli, 2024-09-06:
I still couldn't reproduce this issue 

I have added extra logging for alias handling.

Please install the latest 2.2.2-alpha build and run it in cmd with the command line parameter "-d6", and upload the output( as much as available).

ksio89, 2024-09-07:
@dxli The parameter ``-d6`` doesn't seem to do anything, as nothing is output to the console.

app4soft, 2024-09-07:
> The parameter -d6 doesn't seem to do anything, as nothing is output to the console.

@ksio89 You should add "space" before number: `-d 6`

ksio89, 2024-09-07:
> > The parameter -d6 doesn't seem to do anything, as nothing is output to the console.
> 
> @ksio89 You should add "space" before number: `-d 6`

Still nothing, and I tested on an elevated prompt as well.

lordofbikes, 2024-09-07:
There is no console output on Windows. This only works on Linux or MacOS.

dxli, 2024-09-11:
@ksio89 

I have enabled console debug info for windows for 2.2.2-alpha.

Could you try the latest build in cmd with "librecad.exe -d6"?

ksio89, 2024-09-11:
Here it is @dxli:

[Administrador Prompt de comando - LibreCAD.exe  -d 6.txt](https://github.com/user-attachments/files/16966668/Administrador.Prompt.de.comando.-.LibreCAD.exe.-d.6.txt)

dxli, 2024-09-12:
@ksio89 

Thanks for the crashing logs.

It confirms the crash is not due to alias or block Library handling.

The trouble is we still don't have the exact cause of the crash.

ksio89, 2024-09-14:
@dxli I'm confused because it's not actually crashing anymore after fully uninstalling 2.2.0 before upgrading to either 2.2.1 rc or 2.2.2 alpha. If you wish, I can try to reproduce the issue in a virtual machine, by installing 2.2.0 and then 2.2.1 or 2.2.2 builds.

dxli, 2024-09-14:
That's a great point.

After this fix, we would add a way for auto crash reporting like crash pad in 2.2.2

ksio89, 2024-09-15:
@dxli The crash has reappeared on 2.2.1 rc, so I updated to 2.2.2 alpha to generate a log:

[LibreCAD log.txt](https://github.com/user-attachments/files/17006609/LibreCAD.log.txt)

dxli, 2024-09-15:
Thanks a lot!

This appears to be the menu entry for recent files, which need a code review.

Hopefully, this will be the last blocker for the official 2.2.1 release

ksio89, 2024-09-15:
You're right about the crash being caused by recent files, because deleting all entries on key _`[HKEY_CURRENT_USER\Software\LibreCAD\LibreCAD\RecentFiles]`_ fixes it. Not ideal, but at least I know there's a workaround. On the positive side, I finally found that LC settings are stored on Registry, so I backed up LibreCAD key before uninstalling it.

dxli, 2024-09-16:
@sand1024 
I am reviewing and rewriting some really old code here in 2.2.1.

I feel we need to rewrite in 2.2.2 also.

sand1024, 2026-09-26:
it seems could be closed, can't reproduce under win

closed 2026-09-26 by sand1024