LibreCAD/LibreCAD #1881: Items disappear and sometimes reappear

Questions2000, 2024-09-04:
LibreCAD-2.2.2_alpha1-150-gdc895a29.AppImage
Linux / Debian 12 based distro / GTK based DE

I apologize if this is already reported/known about.

Kind of hard to explain.
With using the newer version of the Continous Release AppImage I on occassion started to notice that items would be drawn, but then when certain tools are applied to them the items end up disappearing and then randomly sometimes come back.  
If you use your selection rectangle and go from right to left to select anything within the selection, you can see that nodes/grips show up but the items are not seen (can also use Ctrl+A select all).
At first when I noticed this I thought they were just random so I didn't make note of each cause, but now I am starting to keep track so not sure if specific to these tools only but, here is an example to reproduce.
They mainly deal with the Divide & Break/Divide tools

`DIVIDE & BREAK/DIVIDE`
`STEPS TO REPRODUCE:`
-Draw a center point circle
-Apply "divide" command
-Click on circle
Sometimes this alone make the circle disappear.
Other times if you click to select the circle and then click points where to divide then the circle will disappear.

Same/similar things occur when using the Break/Divide command.  When done in this case if you hit Ctrl+A to select all it does not appear to show any nodes/grips.

This is a start, just to see if anyone else notices anything simliar to this.  Like I said this may not be specific to these tools.

Thank You

sand1024, 2024-09-04:
hm... quite a weird behavior indeed. I never seen it and not sure I can reproduce it.

The entity disappears from the drawing completely? Or just not drawn properly? What if the the file is saved after entity disappear and then opened again?  And, btw, under which environment (OS etc.) this happens? 

Probably you could make a screencast that illustrates this issue?

sand1024, 2024-09-04:
oh, finally was able to reproduce similar behavior... it seems that the entity is not properly drawn after divide. 
So the divided entity (arc) is still part of drawing, but with some zooms (right after the operation) is is not visible... With changing zoom factor, it becomes visible (and drawn normally after saving and reopening file)... 
weird and nice... I take a deeper look to this.  
Please report if you'll see similar behavior with different operations (in addition to divide) - the most interesting is other entity modification operations

sand1024, 2024-09-17:
disappearing of circle after "Divide" operation - issue related to #1896

sand1024, 2024-10-19:
can you see that issue on latest AppImage? I suppose it should be fixed already and so may be closed

sand1024, 2024-10-26:
@Questions2000 is it fixed already?

Questions2000, 2024-10-27:
I tested this out a couple days ago, was on my list of items to close, because.....
it is fixed :)
Thank you for fixing it

closed 2024-10-27 by Questions2000