LibreCAD/LibreCAD #1882: Add Isometric View tool for projecting drawing

app4soft, 2024-09-05:
### Expected behavior

LibreCAD should has a tool to create isometric projections of selected entities automatically.

Possible solution could be similar to proposed in https://github.com/LibreCAD/LibreCAD/issues/1806#issuecomment-2177890915

> When scaling anisotropically, i.e. differently in xy-directions, explode the polyline and scale the entities separately.

### Observed behavior

Actually LibreCAD has no tool for creating isometric projected view of selected entities.

### Steps to reproduce or sample file

### Operating System and LibreCAD version info

All versions and platforms.

### Reference

- https://qcad.io/rsforum/viewtopic.php?t=8795
  - https://www.qcad.org/doc/qcad/latest/reference/en/scripts/Projection/IsometricProjection/IsoProject/doc/IsoProject_en.html
  - https://www.qcad.org/doc/qcad/3.7.0/reference/en/scripts/Pro/Projection/IsometricProjection/IsoProjectCylindrical/doc/IsoProjectCylindrical_en.html
- https://www.youtube.com/embed/_L5H4nTFxx0

u2fly, 2025-04-21:
From forum, related in part
- https://forum.librecad.org/I-propose-td5726521.html

u2fly, 2025-04-27:
@sand1024

From https://forum.librecad.org/New-feature-idea-perspective-transform-td5726356.html
>  Well, I suppose there could be a good addition there - a support of isometric projections (as predefined defaults for the transformation matrix, of course - so the it will not be necessary for the user to define the matrix).

It would be good to has an option for adding custom projections presets as well.

@dxli 

>Clearly, this new feature contains:
>
>Move,
>Rotate,
>Scale,
>Shear,
>Stretch,
>And perspective.

Shear also needed to enable actually drafted/disabled "Text > Oblique" option ("Text > Width factor" option is Stretch or Scale?).

Also, existing Scale tool has an issue with Arcs if scale factor is negative on any of axis — Arcs scaled, but preserved orientation were should be mirrored. There is no such issue with Arcs in "Text" symbols if "Width factor" is negative (should Mirror tool apply negative "Width factor" for mirrored "Text"?).