LibreCAD/LibreCAD #1884: Window for editing blocks cannot be closed

michael0710, 2024-09-05:
### Expected behavior

If a new window is opened to edit a block, it shall be possible to close this window by clicking on the red cross.

### Observed behavior

A click at the window with the block to edit does not close. It appears as if the red cross just does nothing

### Steps to reproduce or sample file

1. Create an empty file
2. Draw a line
3. Create a block with that line in it
4. Go to the Block List, right click on that block and click on Edit Block
5. In the menu bar, click on Drawings > Arrange > Cascade to be able to see the windows of the actual file and the block
6. Select the window of the block and try to close that window by clicking on the red cross

### Operating System and LibreCAD version info
OS:
Microsoft Windows 11 Pro
10.0.22631 Build 22631

LibreCAD:
Version: 2.2.1_rc3-20-gd7446c30
Compiler: GNU GCC 12.2.0
Compiled on: Aug 31 2024
Qt Version: 5.15.2
Boost Version: 1.75.0
System: Windows 10 Version 2009

app4soft, 2024-09-05:
> It appears as if the red cross just does nothing

If possible, attach screenshots too.

michael0710, 2024-09-05:
The non-working button as described above is the one of the window titled with "Block 'Test'".

One further information related to this issue: Once the red cross is clicked, a message appears in the status bar for roughly 2 seconds saying "Auto-saved drawing"

[image]