LibreCAD/LibreCAD #1895: Drawings is missing entities when exported to PDF through "Export as PDF" menu option

ksio89, 2024-09-16:
### Expected behavior
All drawing entities are visible on a PDF document generated by LibreCAD via **File>Export>Export as PDF**

### Observed behavior
Several drawing entities (like lines, dimension lines and arcs) are not visible in a PDF document generated by LibreCAD via **Export as PDF**, even when layers are set to a line width different than 0 or Default. 

### Additional comments
Entities are visible in a PDF generated by clicking "Print" toolbar button and then choosing a file printer such as Foxit PDF Editor Printer or Microsoft Print to PDF.

### Steps to reproduce or sample file
Export a drawing to PDF via Export as PDF and open it on a PDF document reader application

### Operating System and LibreCAD version info

OS: Windows 11 Home 23H2
LibreCAD Version: 2.2.1_rc3-20-gd7446c30
Compiler: GNU GCC 12.2.0
Compiled on: Aug 31 2024
Qt Version: 5.15.2
Boost Version: 1.75.0

I've attached the drawing and screenshots to illustrate the issue.

[image]

[image]
[sample.zip](https://github.com/user-attachments/files/17016900/sample.zip)

dellus, 2024-09-16:
Those entities missing do have line width 0.00mm.

Options - Current drawing preferences - Dimensions - Extension lines and Dimension lines, arrows and ticks say Width 0.00mm.

Also the lines missing in the drawing itself have Width 0.00mm. Check with Tools - Modify - Properties.

ksio89, 2024-09-17:
The lines have width 0.00m, because I set all layers width to 0.25mm in the drawing and set all entities to use color, width and line type "By layer". Also, the Width value I set in Current drawing preferences>Dimensions but is not saved, as when I open the window again, it reverts to 0.00m for both **Extension** lines and **Dimension lines, arrows and ticks**.

Like I said, the drawing is normal when exported to PDF via PDF printer, it's just the Export as PDF option that is not working correctly.

app4soft, 2024-09-17:
Duplicate of https://github.com/LibreCAD/LibreCAD/issues/1864

@ksio89, you may close actaul issue and continue to discuss in https://github.com/LibreCAD/LibreCAD/issues/186

closed 2024-09-17 by ksio89

dellus, 2024-09-18:
If you want the line width to be as set in the layer attributes, it has to be set as "By Layer".

PDF Printer just handles 0.00mm differently than LibreCAD's Export to PDF, it ties to make a hairline visible even if it is 0.00mm.
Also PDF Viewers can have different behaviour in this respect.

The issue app4soft is refering to does not concern 2.2.1 rc, which you are using, but the later 2.2.2 alpha.

So as there isn't any issue really it's ok it is closed.

Export as image indeed produces only a single width, it's of no use in my opinion. If you need a bitmap file of a drawing better export as PDF and then convert to bitmap in Inkscape or similar.

ksio89, 2024-09-18:
@dellus I set all dimension entities attributes as "By layer" for pen values (color, width and line type), and the dimension lines and extension lines are still rendered as 0.00mm in the PDF document, thus barely visible.

The bug affects 2.2.1 rc as well, that's the version the poster of the issue above stated he/she was using and I tested myself. Just like the user said, the bug is not present on 2.2.0 stable. On 2.2.0, I  exported as PDF the drawing created on 2.2.1 rc, and the dimension lines and extension lines are clearly visible on any PDF reader.

dellus, 2024-09-18:
Sorry, I must have been confused myself. I tested it again, and yes it affects 2.2.1 rc as well. Forget all I have said.
Maybe one beer too much.

Sorry also @ app4soft.