LibreCAD/LibreCAD #1896: Incorrect drawing of Arc if start point and end point of arc are the same

sand1024, 2024-09-17:
### Expected behavior
Arc that is obtained from the circle by divide should be drawn on all zoom levels
### Observed behavior
Arc present in drawing, yet it is drawn only on some zoom levels, and disappears from others. 
### Steps to reproduce or sample file
1) Draw Circle
2) Invoke Modify Divide command on any point of circle - so the circle is transformed to arc
3) try to use different zoom levels

I've found this issue investigating #1881, and while all code of Divide action and RS_Modification seems to be correct, there is suspicious code fragment in RS_PainterQt::drawArc():

  double startAngle = RS_Math::rad2deg(reversed ? a2 : a1);
  double angularLength = RS_Math::rad2deg(RS_Math::getAngleDifference(a1, a2, reversed));
  
  QPainterPath path;
  path.arcMoveTo(circleRect, startAngle);
  path.arcTo(circleRect, startAngle, angularLength);
  
  For the following value:
  a2 = 9.2941394011657135
  a1 = 3.0109540939861272
  reversed = false

The value for angularLength for such type of arc is type is 0 (while actually it should be 360 degrees), and so the arc is not drawn...

### Operating System and LibreCAD version info

sand1024, 2024-09-17:
I'll fix it, just noted it there to use in commit message

sand1024, 2024-09-17:
in Version: 2.2.1_rc3-16-g0a0a81de the arc is drawn correctly

sand1024, 2024-09-17:
well, I think the major question there is whether this is common situation that should be handled in some uniform way or this case with arc is rather exceptional corner situation. 

In later case, I think adding boolean might be overkill, thus so far I've just added the function for checking periodic angles - at least it provides additional clue that angles are actually the same.

Let's see whether this will be enough, at least by quick testing it covers this case. 

So the code is something like this (not committed yet) in rs_painterqt.cpp
.....
`   // brute fix for #1896

    if (std::abs(angularLength) < RS_TOLERANCE_ANGLE) {

        // check whether angles are via period

        if (RS_Math::getPeriodsCount(a1, a2, reversed) != 0) {

            angularLength = 360; // in degrees

        }

    }`

dxli, 2024-09-18:
@sand1024,

Let's commit your fix for 2.2.1.

In 2.2.2, we are going to make the angular length an invariant.

dxli, 2024-09-20:
For 2.2.1, let's try a safe fix: if the start/end points are coincident, assume the angular length is 360 degrees.

For 2.2.2, I feel we should make arc angular length explicit:

Invariant: a2 = a1 + angularLength (or reversed, a1 = a2 + angularLength)

update: angularLength > 2 pi, angularLength = 2 pi for painter,

negative angularLength or more than 2 pi angularLength, not allowed

arc more than 2 pi for example, an arc with angular length 4 pi is not supported. Considered as a 2pi arc (middle point is not coincident with start/end)

sand1024, 2024-10-26:
@dxli I suppose that could be closed as fixed now

closed 2024-10-26 by sand1024