LibreCAD/LibreCAD #1897: Parts of drawing exported to PDF are missing after choosing to automatically fit to page

ksio89, 2024-09-19:
### Expected behavior
The whole drawing is preserved after clicking on "Fit to Page" and exported to PDF

### Observed behavior
Parts of the drawing are not present on PDF after clicking on "Fit to Page". I have to choose a slight ratio so the whole drawing is visible, even when paper margins are set to 0.

### Steps to reproduce or sample file
1. create a drawing
2. on Print Preview, click on "Fit to Page" button to automatically adjust the drawing to the paper size configured and calculate the print scale 
3. export the drawing as PDF
4.  drawing does not fit the page as parts of it are missing.

### Operating System and LibreCAD version info
Version: 2.2.1_rc3-23-g05f8dc7f
Compiler: GNU GCC 12.2.0
Compiled on: Sep 17 2024
Qt Version: 5.15.2
Boost Version: 1.75.0
OS: Windows 11 Version 23H2

Here are some screenshots to illustrate the issue:

1. Print Preview window, showing the drawing at 1:1 scale
2. PDF of a drawing of an A3 page at 1:1 scale, showing that the bottom line of the frame and corner marks were discared
3. same drawing, but at 1:2 scale, the whole drawing was preserved on the PDF document

[image]
[image]
[image]

As you can see, I have to choose a smaller scale (larger denominator) in order for the drawing to be preserved. It seems the PDF exporter functionality is applying non-zero margins of its own after clicking on "Fit to Page" button.

dxli, 2024-09-19:
Can you share the dxf?

ksio89, 2024-09-19:
Here its, sorry for not uploading earlier. Though it happens with any drawing, it's easier to reprodue on a drawing at 1:1 scale. On build 2.2.0.2, the frame is visible, but the corner marks are only partially preserved.

[Legenda A3, paisagem final.zip](https://github.com/user-attachments/files/17067894/Legenda.A3.paisagem.final.zip)

[image]

dellus, 2024-09-20:
For me Legenda A3 exports to PDF completely in LibreCAD 2.2.1 rc23 in Linux appimage.

A Windows issue?

ksio89, 2024-09-20:
Might be, I will test on a Linux virtual machine.

ksio89, 2024-09-21:
Thank you for looking at the issue, but unfortunately parts of the drawing are still missing on latest RC build. The frame bottom line is now present, but the corner marks are still only partly rendered, exactly as on 2.2.0.2. The PDF looks the same as the last screenshot I uploaded, so no need to upload it again. Here is another screenshot comparing print preview window on latest 2.2.1 RC build with the exported PDF of a drawing on an A4 page, also with zero margins:

[image]

And why the console says this after clicking on Fit to Page, even though the paper size is set to A4 dimensions?

``Warning:: Paper size less than 10mm. Paper is too small for fitting to page
Please set paper size by Menu: Edit->Current Drawing Preferences->Pape``

[image]

ksio89, 2024-09-22:
The message is no longer shown, but the margins isuse still not, sadly.

dxli, 2024-09-22:
@ksio89 

this is an actual bug. When printing/exporting (two different things by our implementation), the output needs to follow user drawing settings on paper size and margins.

This bug happens because the way Qt handles paper sizes and margins. Our way was good for Qt4, but since Qt5, our settings became recommendations, not  necessarily  used by Qt.

The current fix makes sure "Export PDF" will be using user settings, so everything should show up (instead of being cropped) in PDF.

If trouble persists in "Printing" (to PDF or not), we will port this fix to printing in the future.

ksio89, 2024-09-23:
But then why is the bug not reproduced on Linux and Windows VMs that I've tested? They use the same version of Qt libraries.

FWIW, the cropping issue is also happening with the LibreCAD installation in my desktop PC running Windows 11. This bug seems completely random if you ask me, and so far seems to not affect virtual machine guests.

dxli, 2024-09-23:
Qt provides a uniform interface for applications across platforms, so mostly, we, "write once, run everywhere".

However, sometimes, like printing, the underneath qt implementations do behavior differently for platforms. Qt printing actually splits into separate systems at quite high level, and leaving us, the qt user to having to split our usages to different platforms also.

For example, printing to files will rely on printer drivers on different OSs, while exporting PDF will rely on qt pdfwriter.

app4soft, 2024-09-25:
> I'm glad to report that this issue has been fixed on latest alpha build (2.2.2_alpha1-172-gb5de5c97), thank you very much!

If so, You, as author of this issue report, may close actual issue thread as "Done".

> Now the only stopping bug AFAIK that needs to be fixed is https://github.com/LibreCAD/LibreCAD/issues/1864, hope it doesn't take too long.

Its probably fixed already, see: https://github.com/LibreCAD/LibreCAD/issues/1864#ref-commit-dba58f3

closed 2024-09-29 by ksio89