LibreCAD/LibreCAD #1898: Block names cannot be reused without reopening the file

Malvineous, 2024-09-20:
If you delete a block, you cannot create another block with the same name until you close and reopen the file.

### Expected behavior

I expected to be able to create a block with the same name as a block I just deleted.

### Observed behavior

An error message appears: `Could not name block.  A block named "example" already exists.`

### Steps to reproduce or sample file

1. Create a block called "example", observe it is added to the "Block List" toolbar.
2. Delete the block from the Block List.
3. Create a block again and give it the same name "example".
4. Observe you cannot create the block because it says the name is in use, when it isn't.

If you close and reopen the file, you can then reuse the name.

### Operating System and LibreCAD version info

LibreCAD 2.2.0-undef
Qt 5.15.15
Boost 1.83.0

lordofbikes, 2024-09-20:
duplicate of #716

Malvineous, 2024-09-20:
Ah my bad, should've searched better, sorry!

closed 2024-09-20 by Malvineous