LibreCAD/LibreCAD #1913: Startup screen image of the build should corresponds LibreCAD version

app4soft, 2024-10-03:
### Expected behavior

Startup screen image should corresponds to the LibreCAD version its starting.

cc @dxli @sand1024

### Observed behavior

I once has been experimenting with setting "Dark" (`librecad01_dark.png`) startup image and it was set to use in `2.2.2_alpha` CI builds with the next workaround:

- https://github.com/LibreCAD/LibreCAD/blob/a03184f4f3dba225e51f0e85e29fef09eb8331a7/.github/workflows/build-all.yml#L41-L48
  - https://github.com/LibreCAD/LibreCAD/blob/master/desktop/media/splash/librecad01_dark.png

With some later commits in `master`, my workaround in `build-all.yml` is now ignored by CI and the latest `2.2.2_alpha` startup screen imag now replaced with a "BETA" (`splash_librecad_alpha.png`) startup screen image in latest builds:
- https://github.com/LibreCAD/LibreCAD/blob/master/librecad/res/main/splash_librecad_beta.png

But just now, the version in `master` was set to `2.2.2 alpha1`:
- https://github.com/LibreCAD/LibreCAD/commit/a03184f4f3dba225e51f0e85e29fef09eb8331a7

> [!NOTE]
>"ALPHA" (`splash_librecad_alpha.png`) splash image exists in repo too:
>
> - https://github.com/LibreCAD/LibreCAD/blob/master/librecad/res/main/splash_librecad_alpha.png

### Steps to reproduce or sample file

[image]

### Operating System and LibreCAD version info

- Version: 2.2.2_alpha1-174-ga03184f4
- Compiler: GNU GCC 9.4.0
- Compiled on: Oct  3 2024
- Qt Version: 6.6.3
- Boost Version: 1.71.0
- System: Debian GNU/Linux 12 (bookworm)

app4soft, 2024-10-08:
@dxli Latest commit solves issue only partially: its correct now for startup splash, but in "About" dialog it still shown wrong.

[image]

[image]

Also, I propose to show in "About" dialog stripped in a half splash as a banner, and here are templates for about banner:

[image]

[image]

And resulted look for "About" dialog of "alpha" build:

[image]

app4soft, 2024-10-08:
And here is comparison of "About" dialogs in `2.2.0.1` (release/stable) and `2.2.2.2` (alpha)

[image]

app4soft, 2024-10-12:
@dxli, this issue almost solved.

The only thing left is to set "About" dialog to fit its minimal size, because actually there are a lot of space in vertical dimension (see screenshots).

### Actual behavior
[image]

### Expected behavior
[image]

sand1024, 2024-10-26:
> The only thing left is to set "About" dialog to fit its minimal size, because actually there are a lot of space in vertical dimension (see screenshots).

Well, It might be that size of the dialog was remembered from previous invocation (with larger logo).

Could you please resize the dialog to required size, close it and open again (of course, my consideration makes sense if appropriate options for persisting dialogs positions/sizes is checked)

app4soft, 2024-10-27:
> Could you

After https://github.com/LibreCAD/LibreCAD/commit/8ed4c62238f9de6ed7cb67ad0a18434f842878b9 there was no issue anymore.

closed 2024-10-27 by app4soft