LibreCAD/LibreCAD #1917: Impossible to change some line types in Grid preferences

app4soft, 2024-10-13:
### Expected behavior

There should not be issues.

### Observed behavior

Actually there are next issues with grids settings:

1. Points Grid — impossible to change the line type for "Meta Grid" (it always returns to _"Dot (tiny)"_);
2. Lines Grid — changing "Grid" line type has no effect (while changing line type for "Meta Grid" works).

### Steps to reproduce or sample file

1. Open "Application Preferences";
2. Change line types for Points Grid and Lines Grid "Grid" and "Meta Grid"
3. Try few settings and observe result.

### Operating System and LibreCAD version info

- Version: 2.2.2_alpha1-180-g8ed4c622
- Compiler: GNU GCC 9.4.0
- Compiled on: Oct 12 2024
- Qt Version: 6.6.3
- Boost Version: 1.71.0
- System: Debian GNU/Linux 12 (bookworm)

sand1024, 2024-10-13:
yes, I can confirm this bug... just a quick "last minute change" issue, will be fixed in next update.

app4soft, 2024-10-17:
With latest build its better, but there is an issue with setting "Dots" linetype for "Metagrid" in "Points" grid settings: now there is drifted offsets betwen grid and metagrid points that looks buggy

[image]

Expecting that there should be additional "Grid type: Classic", withe the classic pooint grid settings.

### Verion info

- Version: 2.2.2_alpha1-181-g8b3cf9ca
- Compiler: GNU GCC 9.4.0
- Compiled on: Oct 17 2024
- Qt Version: 6.6.3
- Boost Version: 1.71.0
- System: Debian GNU/Linux 12 (bookworm)

sand1024, 2024-10-17:
hm... well, it's hardly possible that the last one may be fixed. Dot in meta-grid are just pattern for the line used to draw meta grid, and they are not related somehow to the grid metrics.

Of course, potentially it is possible to draw metagrid also using individual points instead of lines, however, the logic there is not too obvious, as for me, for the end user. 

If we'll draw metagrid by points, it seems that that points should be drawn in the same positions where grid points are located. 
So the only difference will be just color of points for metagrid and that's hardly will be too visible. 

I'll think about this, yet can't promise that it will be supported.

app4soft, 2024-10-17:
Actually there are two tabs for Points and Lines grid types, with 2 sub-tabs of settings for these types.

I propose to unify grid settings and keep only one grid sub-tab — this would removes extra grid type switcher and one extra sub-tab, see:

[image]

Also, I'm suggesting there should be a simplification: "Line type" should be renamed to "Type", and drop list options should consists of three entries:
1. Points (as it defined)
2. Lines (aka continious lines)
3. Custom — for this option if enbaled there should be a text input line shown where user could input line type code using AutoCAD Linetype Definition syntax, see:
- https://github.com/LibreCAD/LibreCAD/issues/1738#issuecomment-2310528944

sand1024, 2024-10-19:
> I propose to unify grid settings and keep only one grid sub-tab — this would removes extra grid type switcher and one extra sub-tab, see:

Well, these two tabs were made in intentionally - actually, colors for grids with points and for grids with lines  should be different. Initially, there was only one setting - yet I've found it annoying to update color settings when changing between points and lines. The issue that the same color combination is fine for grid point and ugly for lines grid (and vise versa). 

Of course, it's hardly possible that the user will change grid type too often - yet still... 

> Custom — for this option if enbaled there should be a text input line shown where user could input line type code using AutoCAD Linetype Definition syntax, see:

Well, maybe in some future. At the moment, custom user-defined line types are not supported. 

While it's quite simple to support different combinations of points and dots within line, proper support of more complex line types (say, zigzags, curves, or even just text inside a line) is a really big task that requires lot's of development efforts. 

At the moment, existing line types are nicely mapped to capabilities of custom line types offered by QT. 

However, more complex  polyline-like compound line types will require creation of heavyweight painting logic. For simple cases like line - that's easy for sure, yet think about more complex paths (like ellipse, spline or even just circle).  

For example, for [https://github.com/tuffs/versacad-to-autocad-lines/blob/master/acad.lin -](https://github.com/tuffs/versacad-to-autocad-lines/blob/master/acad.lin) you've mentioned - all line types below  `;;  Complex linetypes` section are quite challenging to implement. 

Another challenge there it to make line types portable enough so they will be supported by drawings created by other software (and vise versa). For example, AutoCAD line types also supports shapes from .shx file - it means that we have to add proper processing of it too. 

Of course, support of custom line types is quite demanding feature and support of it is planned, yet not sure it will be added really soon.  

Returning to the file with line types above - potentially we'll add support for linetypes in form like one defined from the start of the file till complex line types first, let's see..

u2fly, 2025-04-14:
> While it's quite simple to support different combinations of points and dots within line, proper support of more complex line types (say, zigzags, curves, or even just text inside a line) is a really big task that requires lot's of development efforts.
>
> Returning to the file with line types above - potentially we'll add support for linetypes in form like one defined from the start of the file till complex line types first, let's see..

@sand1024 Adding LIN-file support with simple "points and dots within line" would be enoug to fix actual issue with ISO standard linetypes (_instead of hardcoding each missed ISO standard linetype_):

- https://github.com/LibreCAD/LibreCAD/issues/1738#issuecomment-2800137636

Closing this issue in favour of initial LIN-file support request.

closed 2025-04-14 by u2fly