LibreCAD/LibreCAD #1918: Break Divide tool does not recognize circles when using tangent lines

Questions2000, 2024-10-13:
Linux / Appimage alpha

First off, the break/divide tool that has been added is amazing!
I finally got around to testing it out a little and really cannot believe it, I really want to say a lot more about how great this tool is but want to keep this post as short as possible :)  just thank you very much for adding this. 

For now I just want to mention something that I noticed, that may already be known, I apologize if it is.
I noticed that when using "tangent line" based tools with circles that the break/divide tool does not recognize the pieces of some of the circles.

I first noticed this when using the "tangent line" tool that adds a predetermined tangent line to a circle.

I then noticed this with "tangent line" tool C to C which I will share example of how to reproduce.

`STEPS TO REPRODUCE`
-draw two circles one smaller than the other, side by side with some space in between
-use the tangent line C to C tool to draw two tangent lines, one from the top of one circle to the other, then other line from bottom of circle to the other circle.  So now that appear to form one combined shape.
-then use break/divide tool to try and cut the inner parts of the circles between the tangent lines out.
You will noticed that the first circle will allow the center piece to be removed but then the second circle does not recognize the center piece, only the full circle itself.

Hope this helps
Thank You again for adding this tool.
 I also have an idea to share about it but I will make that post in the future.

Questions2000, 2024-10-15:
@dxli
Thank you for looking into and thumbs up

@sand1024 
Thank you for looking into it and again for creating this tool.
I figured it had something to do with how the lines connected with tangent lines.
What confused me though is that the first circles center section does actually trim out and delete, so it is strange that when you try it with the other circle which is in essence the same action, then it does not work (hopefully this makes sense).

Thanks again

sand1024, 2024-10-15:
could you please make a small drawing with this case and share it there? 
Probably it may be related to angles and radiuses values, so actual example will be helpful.

lordofbikes, 2024-10-15:
Probably related:
#523
https://forum.librecad.org/cannot-snap-intersection-of-two-inscribed-ellipses-td5725436.html

And there are probably more threads and issues addressing this.

Recent [post](https://forum.librecad.org/cannot-snap-intersection-of-two-inscribed-ellipses-td5725436.html) from @dxli hit the bulls eye.

Another workaround is to create a temporary orthogonal line through the circle center and use its intersection. There is much less room for tolerance issues with orthogonal intersections.

closed 2024-10-15 by lordofbikes

Questions2000, 2024-10-17:
@sand1024 
Here is a good example, I put some time into this post already, but lets say I went even further and filled in all the categories in the template and provided images, etc. would it have been worth it?
The conversation was with you and the tool you have provided, you sounded like it might be something you might look into in the future.

The link that was shown is to the forum so there might not be a ticket here on Github to look into the issue because this post was closed, based on probablies?
 "Not Planned", maybe not by our lord, but maybe by you or other developers, so why not keep this open?  Again I am not a developer so I have no idea, but seems strange.
Does the tool you have provided have any connection to the link that was given?

I also have mentioned that it does work for one of the circles, so would think it would be worth keeping this post open for a little while to look into it, rather than closing?
I can replicate it.

So what should I do?
Any chance this could be reopened, if not just to let the people having the discussion, finish it and decide to close it after?
Let me know

Questions2000, 2024-11-10:
@sand1024
Sorry to tag you but because this issue has been closed just wanted to make sure this was seen.
Not sure if this has been worked on, but in testing this, it seems to work a lot more then it did all of sudden.

In testing have not been able to generate any instances where neither circle can be trimmed
and
only once did I get it to generate a scenario where only one circle accepts the trim.

I was just wondering if this was worked on, if not, then something happened which made this work better.

If this was worked on by you (or any other developers), just wanted to let you know that I noticed and thank you for updating this, I appreciate it.

sand1024, 2024-11-11:
@Questions2000 
Yes, internal algorithm for trimming circles was improved, and as side effect of such improvement "Break/Divide" command behaves better now. 

YET! if you see that there is still a scenario with invalid cut of circle - please report it as your did it before (with specific sizes and locations) or please just attach example dbf file with such case.