LibreCAD/LibreCAD #1919: Many issues detected with new continuous release AppImage

Questions2000, 2024-10-15:
LibreCAD-2.2.2_alpha1-180-g8ed4c622.AppImage
Linux / Debian 12 based distro / GTK based DE

I was not sure if developers regulary check the "Discussions" section so I thought it was important to put this question here, hope this is okay.

I just started testing out the following AppImage version
`LibreCAD-2.2.2_alpha1-180-g8ed4c622.AppImage`
and have noticed that many issues occur.

Not sure if they are expected / known about issues but they are with common items that have all worked in previous versions but now do not.
Should I post each one separately? There will be many posts, so before I do I just wanted to check in with everyone.

Basic overview of them....
-Grids do not show up
-Snaps do not work
-Mouse hover over items do not highlight
-Can no longer drag items to move them with mouse where it moves from the closest grip point, only circle center points work.
-Etc.

Just was not sure if all these are related to something similar for them all to stop working all of a sudden.
If someone could let me know how to proceed.
Also if any other users can verify some of these just to let developers know if it is just me.
Thank You

sand1024, 2024-10-15:
well, I suppose I've mentioned this previously - please report each issue individually, one issue per ticket. 
ideally, with steps to reproduce and sample files.

lordofbikes, 2024-10-15:
> well, I suppose I've mentioned this previously - please report each issue individually, one issue per ticket. ideally, with steps to reproduce and sample files.

That's why we have an issue template! The reporter should think about his report and put some effort in it, give clear description and probably add a screenshot(s) to clarify the request. Otherwise the developers have to suppose and investigate and check back, what costs time that could be used better.
And the template works for feature requests too.

Our GitHub Discussion section was just recently reopened for you, @Questions2000, because many of your posts are just discussions. We have https://forum.librecad.org/ where developers and experienced users, much more people, who take part in discussions and can support you and clarify or avoid one or the other issue here.

So please try this and respect the issue template, don't delete it.

closed 2024-10-15 by lordofbikes

Questions2000, 2024-10-15:
Did anyone read my post?
I only made this post this way because I was concerned that there might have been some big overall issue that occurred with the last release because all of a sudden many things that worked before were not working all at the same time.  I just was checking in before I made a ton of separate posts reporting each issue separately.  As mentioned I just wasn't sure if any of the developers went into the "Discussions" section, the only developers name I have seen show up there is "dxli" no sand1024, no lordofbikes, etc.

As far as the Discussions comment, I have only had two posts of mine that were MOVED to the "Discussions" section and they were made as you said, before the "Discussions" section was even available to post in.
One was a list of "Settings" that sand1024 mentioned to share and the other was a feature request to offer a setting to turn off the grid.
How is that any different then any other users feature requests?
All the other feature requests in the issues section are discussed.
And again both were made before the "Discussions" section was available, so not sure why the upset response?  If you were making a feature requests or bug reports for a different project, would you want to discuss/debate with other users or the actual developers that actually have something to do with whether or not that feature will be added or that bug will be fixed?

As far as the template, yes I apologize I will start to use, often deleted because it contained a commented out section, but I put my info at the top instead of at the bottom to let the developer know what version using before anything else because I feel it is the most important, and I generally cover all the other three topics requested, but even that not everyone else uses the bottom section and nothing is said to them.

Just understand that I am just trying to help, I spend a lot of time testing and sharing what I have found because I think the program is great and worth the effort for me to do so, apologize for the issue it caused here based on the responses.

Again I explained in my original post, I only did this for a specific reason and I was going to close this post after I received a response, did not think it would be as big of a deal as it turned out to be, sorry.

sand1024, 2024-10-16:
Well, I suppose the major point of @lordofbikes  was a bit different. 

The case is quite simple - in order to fix something, developer had to understand what to fix.  

And ideally, that information should be precise - simply because LibreCAD is quite a complex software, plus it heavily depends on the data stored in drawing. Plus it operates under different operation system... 

Therefore, even if you see some issue on your machine - it might be that under other operating system and in other drawing that issue will not occur.  That's the reason why precise description is needed for the developer. 

Another point - is how to handle issues you've found efficiently.  The golden rule here is quite straightforward: one problem - one ticket.  Imagine that, for example, you've reported two issues in the same ticked - yet one is severe bug that should be fixed and another one is just a "nice to have" feature. It's a kind of nightmare for the project to process such cases and see the overall issue with bugs clearly. 

In general, regressions and errors in AppImage is for sure quite bad thing - however, the latest version is development snapshot that is under active development. Of course, it's not as stable as release version. 

Thus you feedback is more than valuable for fixing all that issues - the only thing that we need to understand precisely what is broken and what should be fixed. 

So returning to you initial post - the better way to report it is to create a set of tickets with details, like 

Ticket 1: Grids do not show up
Ticket 2: Snaps do not work
Ticket 3: Mouse hover over items do not highlight
Ticket 4: Can no longer drag items to move them with mouse where it moves from the closest grip point, only circle center points work.

**-Etc. -** no way we'll be able to fix Etc. :)

Questions2000, 2024-10-16:
@sand1024 
Thank you for your reply.
Not sure why all of a sudden what feels like a teaming up is occurring?
I have no issue with what you guys are saying, for the most part my posts have contained one issue have they not?  The main one that did not was a list of "Application Preferences" which you mentioned it was okay to make, although there were a lot of items mentioned in the post, the idea was that they were all mentioned under the topic of "Application Preferences" which you mentioned you were reorganizing which was the only reason I made the post.  You did not even respond to that post, did you see it after it was instantly moved into the Discussions section?
In that post I also mentioned I was going to make separate posts for the items that were new features, I started with the first post about grid feature and that was instantly moved to the "Discussions" section.  

Most feature requests on this site, end up getting "discussed" in the comments, that is part of the point of a feature request.  Speaking of this, after all that about defending the "Discussions" section and that I should post there, I just noticed that that section is now no longer available and the posts I have made have all been closed?

Because your response still went back to this view, I will say again, I just made this post here the way that I did because all of a sudden I just noticed many issues all occurring at once and it just felt maybe there was possibly a bigger issue.
I was just looking for a simple response of something like.... "yes we found an error in the compiling which caused some issues" or "no all these issues are separate" and right after that I was going to close post.  It was just a broad question to see if there was a known about cause or not, that is all.
It was just meant to be a heads up to you guys just in case it was a larger issue that may have been causing problems.  Not sure why everyone is still approaching this in this way?
All three of you, at the same time no less.
My post regarding command line height #1805, everyone here should read that post all the way through, interested to hear if anyone notices anything.

I thought the time I was putting into this / my posts were helpful, at least this was my intention, now I am not sure what to do, does not feel that good.

I will probably make the appropriate modifications to my posts.
Thanks again guys

LABELS QUESTION
This post has already gone into a different direction then was intended and also has been closed, this is on topic to the current topic we are discussing, just a quick question for you or anyone else.
Labels, the colored labels that are added to posts here, is it possible for average users like me to be able to add them to my posts or is this only by moderators, developers, etc. ?

sand1024, 2024-10-17:
well, without any doubts - your time and posts are really helpful.  
The only thing that for developers need detailed and structured reports for efficient processing them. 

As for feedback, like for LibreCAD#1805 - as soon as post is created, it not always requires some additional communication - some issues are either obvious, or simply do not require some clarifications or feedback. 

So please be sure that even if no one responded to you post, it does not mean that it is not checked.

lordofbikes, 2024-10-17:
I'm on the run for a trip, so just time a few words to avoid further escalation.

@sand1024 already stated, that your contribution is much appreciated, my opinion too, so keep it up.
@sand1024's hints about reporting single issues, compact and informative are on the point.

The issue template suggests to delete the comment section, not the headlines. They show what the developers need and the sections should be filled with content by the reporter.

I have no numbers, at least that doesn't matter, but I think there are regularly issues, where you mention doubts if this is the right place or if suggestions are valuable. In such cases, the forum is our preferred place, where other users can help to clarify doubts and work out feature requests. E.g. discussing with other users can join multiple opinions in a feature request and save developer time. Then issues or feature requests can be reported clearer.
When discussions are started in issues, most participants are developers, who can have a different view on features than users. Vague issues can become boring and people might give up reading, before they come to a valuable point.

It was not my intention to offend you, just a hint to use the forum to clarify your doubts. This will make your issues more efficient.

Questions2000, 2024-10-18:
Thank you for the replies.

@lordofbikes
Thank you for taking the time to make this post before your trip, I really appreciate it.  Hope you have a safe trip and if the point of the trip is to have a good time, hopefully you have a good time :)

I do not know why all the issues occurred specifically, but I figured out what caused them in general.  I tried clearing out my .config/LibreCAD and then running the newer LC AppImage and now these items work again, for the most part.
I might have found some minor issues with grid but I think another user already reported the issues, but the grid does show up now.  With snaps that were not working at all, are now working but have had a couple moments where they randomly have not but not sure yet, but overall they work again.  I will report anything I can find.
My guess is that because there were some settings changes made to LC that older .conf files are not as compatible with newer versions of LC.  Developers who read this, just keep this in mind if any other users mention anything similar to what I have mentioned, I have also made a note of it in the original issue post at the top.