LibreCAD/LibreCAD #1920: Check for new versions at startup still tries to connect even when setting is off

Questions2000, 2024-10-15:
I do not like when applications overly try to connect to internet without my permission.
It appears that LC now has a setting that checks for new versions at startup.
I would not have known about this if it was not for the following dialog message that appeared
"Sorry, some network error occurred during checking for new version"

When I went into the settings and saw the following setting
“Check for new versions on startup”
I tried unchecking this option to turn it off.
Even though it is unchecked, it still appears to be trying to connect to internet without permission because the following error still shows up
"Sorry, some network error occurred during checking for new version"

Not sure if I am not doing something correctly?
If someone could take a look at this.
Thank You

Questions2000, 2024-10-27:
Are you saying that something will be changing in the next update?
I tried this version `LibreCAD-2.2.2_alpha1-190-gdfecd0ed.AppImage` and this error dialog box still occurred even though 
`Application Preferences - Defaults - New Versions - check for new versions on startup` is unchecked
I like the manual option you mentioned in the About dialog, I overlooked that one.
Let me know and maybe I can test this out.
Thank You

sand1024, 2024-10-27:
hmm... that's really strange, and no, my previous comment was for previous alpha, in 190/193 I supposed that should be fixed... 
Ok, I'll check again.

Questions2000, 2024-11-10:
Okay, just letting you know that I tried this in the latest AppImage version just to be sure and this still occurs.
`LibreCAD-2.2.2_alpha1-198-g8919ed83.AppImage`

sand1024, 2024-11-29:
@Questions2000 - this should be fixed in last AppImage, btw

Questions2000, 2024-11-29:
I had just tested this yesterday and was going to mention that I noticed this seemed to be fixed.
Thank you for fixing this.

closed 2024-11-29 by Questions2000