LibreCAD/LibreCAD #1921: Set Layer Tree new layer dialog to open with focus in the name field

Questions2000, 2024-10-15:
If possible, set the Layer Tree new layer dialog to open with focus given in the name field automatically so user can instantly type in new layers name.  I believe this is how the Layer List widget used to function.

Thank You

Questions2000, 2024-10-18:
@sand1024 
I tested this and it appears to be updated if you want to close this?
Thank you for making this update.

sand1024, 2024-10-18:
yes, this is fixed in last update, so the issues may be closed. 
This one as well https://github.com/LibreCAD/LibreCAD/issues/1912
There is a button in toolbox near to draft button and it is used disabling screen width of lines, named "Draft Lines"

Questions2000, 2024-10-18:
Oohhh I didn't know about that one, thank you for updating that as well :)

closed 2024-10-18 by Questions2000