LibreCAD/LibreCAD #1923: Alpha builds are crashing after selecting Hatch

ksio89, 2024-10-16:
### Expected behavior
Application doesn't crash after selecting Hatch feature

### Observed behavior
Application crashes after selecting Hatch feature

### Steps to reproduce or sample file
Create a closed entity and select Hatch feature (either by clicking on toolbar button or by typing the command), and the application will hang on "Choose Hatch Attributes" window for a while and then crash. I've attached a screenshot and the console output just before application freezes.

### Operating System and LibreCAD version info

Version: 2.2.2_alpha1-180-g8ed4c622
Compiler: GNU GCC 11.2.0
Compiled on: Oct 12 2024
Qt Version: 6.6.3
Boost Version: 1.83.0
System: Windows 11 Version 24H2

[image]

[LibreCAD.txt](https://github.com/user-attachments/files/17399627/LibreCAD.txt)

sand1024, 2024-10-16:
hm.. potentially, it might be that some hatch data is corrupted. Could you please create archive of hatches stored in hatches directory (you can refer to Application Preferences->Paths) and attach it?

ksio89, 2024-10-17:
I didn't set a default directory for hatches or use custom hatches, but I uploaded ..\LibreCAD\resources\patterns folder in case that's what you're looking for.

[patterns.zip](https://github.com/user-attachments/files/17412078/patterns.zip)

sand1024, 2024-10-17:
thank you for uploading the archive.
It's strange, yet I can't reproduce the issue with the latest AppImage (2.2.2_alpha-latest).   
Actually, there were changes related to hatches, so it might be that original issue was gone with last update. 
Could you please update and check again?

ksio89, 2024-10-20:
Yep, it has been fixed in the latest alpha build, thank you guys.

closed 2024-10-20 by ksio89