LibreCAD/LibreCAD #1926: Showing double statusbars by default

Questions2000, 2024-10-18:
POSSIBLE GUI UPDATE

### Expected Behavior
To see only one status bar by default.
### Observed Behavior
Default setting appears to show a blank “statusbar" at the very bottom of screen.
The statusbar with coordinate values, icons, etc. is right above it.  
So it looks like LC is showing two status bars at the same time, one standard and one blank below it.
### Steps To Reproduce
Open up LibreCAD with default settings and this occurs.

I checked in the Preferences and the only thing I see about statusbars is
Preferences - Defaults - Startup - Use “classic” statusbar
this option is off (no check)

### Operating System and LibreCAD Version
LibreCAD-2.2.2_alpha1-181-g8b3cf9ca.AppImage
Linux - Debian 12 based distro
GTK based Desktop Environment

sand1024, 2024-10-18:
well, so far this is intentional. The bottom status bar (if it's visible, since it's possible to disable it via popup menu) is used for showing some notification, but probably in the future the functionality of it will be extended (say, by showing progress, or prompt or something else). 
If it's annoying or waste of the screen space, it's better to hide the bottom status bar (of course, this is possible only if status bar is not in old "classical" mode, i.e. not solid one).

Questions2000, 2024-10-21:
Okay thank you, I didn't realize that this blank bar was a new status bar that showed information.
I will close.

closed 2024-10-21 by Questions2000