LibreCAD/LibreCAD #1927: Statusbar loads up in the snaps toolbar by default

Questions2000, 2024-10-18:
### Expected Behavior
To have snaps toolbar appear above the statusbar by default as I believe it used, not both combined in the same toolbar by default as it now does currently.  
I know user can just move the items around the way they like but seems like the statusbar (even though it appears to be made more toolbar like) should still appear at bottom the way it did, for the default LC setting.

### Observed Behavior
Default layout appears to now show the statusbar with all its content in the toolbar that used to contain snaps tools.
Now the snaps toolbar shows up to the right of the statusbar, all in the same toolbar (row)

### Steps To Reproduce
I removed all the content from my .config/LibreCAD folder.
Open up LibreCAD with default settings and this occurs.

### Operating System and LibreCAD Version
LibreCAD-2.2.2_alpha1-181-g8b3cf9ca.AppImage
Linux - Debian 12 based distro
GTK based Desktop Environment

sand1024, 2024-10-18:
oh, that's good point - I'll check how status bar behaves on default/initial settings. 

And you right - now components of status bar are first class toolbars - so they might be mixed with other toolbars. 

Actually, the major point for this is simple - the intent is to make UI more compact and give more screen space to graphic view (after all, drawing area is more important than tools around). 

Of course, it depends on screen resolutions - yet on larger screens, it's quite convenient to use single bottom row for snaps, status bar and probably some other tool.  For screens with small resolutions - yes, such layout is an issue.  

So anyway, I suppose that it seems that most of users still have to adjust layout of their workspace ...

Questions2000, 2024-10-21:
I understand some users might prefer one toolbar and other users might prefer the way that it was where they are stacked with the statusbar below at the bottom.
I will keep this open until you developers decide which should be the default.
If you guys decide upon the use of all in one toolbar as it currently is, then this can be closed.
Thank You

Questions2000, 2024-10-27:
Let me know about this one, if you would like me to close.
I opened up older versions of LC and because the classic statusbar was at the bottom below the toolbar, I just figured that this would be the default starting location of the new statusbar toolbar, but not sure how you guys feel about it.

Questions2000, 2024-11-29:
Just decided to close this issue.

closed 2024-11-29 by Questions2000