LibreCAD/LibreCAD #1928: Change color icons to color boxes in Preferences

Questions2000, 2024-10-18:
POSSIBLE GUI UPDATE

Preferences now shows the same icon/button to change items colors which is good for a consistent look.
However, if it does not look too busy, I wonder if it would be more beneficial to show a square with that items current color instead of this default standard icon for colors.  Color codes are not typically human readable and these icons showing the actual colors currently being used may help users find what they are looking for visually if they are not familiar with the settings themselves (they can just find the item to change based on its color).
If this is added, I would put a black outline around the color just in case colors are similar to the Preferences background color.
The current color icons are newly added so it is okay if do not want to change them and again they are consistent.
I do not have any problem if this issue is closed, it was just an observation.

Thank You

sand1024, 2024-10-18:
hmm... well, in general this is quite productive idea to provide user with indication of actually used color there... the only thing, that using the color-filled button may lead to visual overloading of dialog's UI by different colors. Plus, it's still convenient for some users to enter color quickly by code (instead of selecting it via dialog). 
I'll keep that idea in mind, anyway.

Questions2000, 2024-10-21:
@sand1024 
Thank you for your reply.
No I am not saying you would get rid of the field where user can enter a color code, that would remain.
I am saying that just the current default icon (to load that color picker dialog) would be replaced with an icon that is a square or rectangle icon filled with the current color selected, just to give user a visual of what the color is selected.
The button itself and the field to type in a color name or color code would all remain as is.
Thank You

closed 2024-10-27 by Questions2000