LibreCAD/LibreCAD #1929: Change the snap shape used for middle points

Questions2000, 2024-10-18:
GRAPHICAL UPDATE

To me it seems that all the other snap points can get away with being the same shape.  However, I feel that “middle point” snaps could benefit from showing up as a different shape then all the others, because it is a snap point that is usually not really visually known.  Meaning at times can kind of make you question if you are actually snapping to the middle point of an item or not.
This increases especially in some situations where there are items overlapping or if you happen to have both “Entity” snap and “Middle Point” snap on at the same time.
If this was added would obviously be important that the snap shapes shift smoothly between each other and are not overly visually distracting to the user, which would make this not worth adding.

Developer would have to decide....
-Should this be an option that can be turned on/off?
-Should user be able to decide the shape to use?
-If not, then what shape should be used as the default?

Being that square and circle shapes are already used, I would probably use a Triangle.

Thank You

Questions2000, 2024-10-25:
If not interested in changing the shape of the middle points snap, could also be offered as maybe a change of color of the snap which maybe is easier to program.

sand1024, 2026-09-26:
@Questions2000 
Just re-read the issue and not sure what is expected there - changing the color of snap indicator based on snap point type? 
So far, the type of possible snap is shown in informational cursor, but probably you're suggest adding some marker in snap point candidate? Please expand on.