LibreCAD/LibreCAD #1931: When new statusbar is turned off the setting does not save

Questions2000, 2024-10-21:
### Expected Behavior
If the new statusbar (blank bar at bottom) is turned off, to save/remember this setting for the next time LC is launched.

### Observed Behavior
When you turn off this new statusbar, the setting does not appear to save and it shows up every time LC is launched again.

### Steps To Reproduce
-Legacy statusbar setting is set to off (unchecked)
-Right click on any toolbar to open context menu
-Uncheck "Statusbar" to hide the blank statusbar at the bottom of screen.
-Close LC completely
-Launch new LC
-Notice at the bottom the new statusbar appears again, the state of it does not appear to be saved.

### Operating System and LibreCAD Version
LibreCAD-2.2.2_alpha1-181-g8b3cf9ca.AppImage
Linux - Debian 12 based distro
GTK based Desktop Environment

sand1024, 2024-10-26:
included into last alpha, may be closed

Questions2000, 2024-10-27:
I tested this out yesterday and confirm that this works great, thank you for this update.

closed 2024-10-27 by Questions2000