LibreCAD/LibreCAD #1932: Add the color orange to the layer color drop down list

Questions2000, 2024-10-21:
### Expected Behavior
To have all the colors of the rainbow

I see that all the other colors have a color and then a “Dark” version of that color right after.
If this is added would probably follow same pattern.
Orange
Dark Orange

Unless there is some reason why it is not included?
I think this is a legacy thing with computers in general of not including the color orange but have always found it kind of odd that some programs still continue this.
Not important, if developers are not interested and want to close.

### Observed Behavior
No color orange in the color list

### Steps To Reproduce
Go into Layers Properties and click on “Color:” drop down to view color list

sand1024, 2024-10-24:
yes, there is a reason for this - there is no predefined "Orange" color in QT and it seems it's legacy thing indeed. 
Of course, it's easy to add some color closed to orange, yet it selectin will be a kind of personal preference

These are good comments regarding color names 
https://doc.qt.io/qt-6/qcolorconstants.html

However, I think it might be a good idea to add orange/dark orange

Questions2000, 2024-10-25:
@sand1024 
Thank you for your reply.
I made the following post before I realized that anyone replied.
I was not aware that this was based on QT itself so I viewed getting these color codes differently.

I went into one of the color fields in “Application Preferences” and typed in “Orange” which actually worked.
This is the color code it uses for Orange `#ffa500` so this could be the “standard” color code used for this color?
For the Dark Orange color code I just chose was just pulling down the vertical color fader on the right side of the color picker dialog to about 50% like all the other standard colors seemed to do to get their Dark versions.

ORANGE
#ffa500
DARK ORANGE
#c27b00

closed 2024-10-29 by Questions2000