LibreCAD/LibreCAD #1934: Keyboard Input for Tool Options

GSStnb, 2024-10-21:
### Expected behavior
Enhancement request: Add keyboard input to options available in toolbar.

### Observed behavior
Line - Angle, Line - Vertical and Line Horizontal each have two common options; length and start point.  These options need to be changed on the option toolbar, necessitating navigating from the drawing area to the toolbar and back.  There is no command line prompt as there is, for example, with offset distance for parallel.

### Steps to reproduce or sample file
Pick one of the three tools to draw a line of a given length and from a snap point.

Add a command prompt for length and snap point (B | M | E).  A prompt for angle, of line - angle,  is also needed.

### Operating System and LibreCAD version info
(All)

sand1024, 2024-10-22:
well, it looks like so - but that's not actually true. 

Please start line->angle command. In command line (where it's asks for position) - type "help". 
You'll see the list of expected commands. 
You can type "angle" - and enter angle. "length" - and enter length. 

"help" command is general command that is supported by all actions and it displays additional commands (which depends on the current state of the command, btw) that could be entered via command widget for the command customization.  

Of course, the command prompt may be misleading, but here is a kind of tradeoff between screen space and informational message. 

Anyway, I suppose command widget will be improved in next updates.

GSStnb, 2024-10-22:
I tried that and yes it does work that way, but why?  It isn't intuitive, quicker or consistent with other tools.  If, for example, I want to create an arc I am prompted for each attribute; centre, radius, etc.  Why not the same for a angled / vertical / horizontal line?  Select a tool and then enter the attributres at the command; (angle), length and snap point, e.g. la {ENTER] 45 [ENTER], 50 {ENTER] and B | M | E [ENTER].  Currently it would be; la [ENTER], angle [ENTER], 45 [ENTER], length [ENTER], 50, [ENTER] and then still to the tool option toolbar to select Start | Middle | End.

sand1024, 2024-10-22:
hm.. well, it's a good point - and it even quite valid for that specific command. 

However, please note that most of entered values are saved for later use.  Therefore, they are rather optional... 

For example, if you enter angle/length, next line will be drawn using previously entered angle and length (so only the position of the line is required). 

In addition, some tools has quite a long list of command - and thus forcing the user to enter all of them one by one will just lead to forcing the user to press enter many times (if the user is ok with defaults). 

So far the logic of most drawing actions is as following: 

Required values are asked to be entered by prompt (and they may depend on or affect the current state of action). 
As for optional attributes - it is expected that they should should indicate that it is desired to enter them by entering text command in the command line.  Thus, for line angled - position is required value, while angle and length - optional (as they are set to value entered during previous invocation of the action).

Yet anyway, I feel that the operations of command widget should be improved - so please don't hesitate to share any ideas you might think may be convenient. 

Snap point in option toolbar - well, yes, that's true - support of setting it from the command line should be added.

GSStnb, 2024-10-22:
The 'Parallel' tool is an example of a tool that works well in both the drawing window and the Command line.  After selecting the tool, the command prompt is "Specify Distance <10> or select entity or [through]".  So I can either enter a new distance or accept the current stored value and just select an entity.  

Perhaps an event in the drawing window can bypass any further prompts - but I need to run through all the tools to see if that is the case or even makes sense.  I can think of another off the top of my head - circle with radius, but I'll start on a list...

GSStnb, 2024-11-16:
I'm happy to, but I have several things on the go.  I'll continue to work on it when I can.