LibreCAD/LibreCAD #1936: "Line Join" should not create new lines

gerardbm, 2024-10-23:
The "Line Join" function is really useful, so I love it has been implemented in LibreCAD. This one should be improved to achieve the expected result.

### Expected behavior

The two lines selected should be joined, without creating two new lines. We start with two lines short and it should end with two lines longer.

### Observed behavior

When two lines are joined, in fact there are two new lines created. We start with two lines short and it ends with two lines short and 2 lines longer.

sand1024, 2024-10-24:
hm... actually, there is a policy for each line in tool options bar. Please check them. 

If you select "trim/extend" value - the line will be extended (or trimmed by intersection point). 
if you select "add segment" - additional line from endpoint to intersection point

And, in addition to that, there is a setting that specifies that original line(s) should be kept or removed. 

So please play with options in tool options toolbar and let me know if behavior of the command does not correspond settings (and so it's really a bug). Otherwise - such behavior may be valid according to settings, if it so - please close the ticket.

gerardbm, 2024-10-24:
That's perfect! I didn't see this option in the tool bar. Knowing that, I only need to enable the checkbox to remove the original lines, and then it stays checked. Great and awesome!

gerardbm, 2024-10-24:
That's perfect!

closed 2024-10-24 by gerardbm

sand1024, 2024-10-24:
Thank you :) 

Just a side note - there are lots of options available via options toolbar which is action-context aware (lie, division mode, offsets, generic parameters and so) 

So for productive usage, be sure that you aware of options toolbar and know how to use settings from there for specific actions.