LibreCAD/LibreCAD #1937: "Break/Divide" should also remove independent segments

gerardbm, 2024-10-23:
The "Break/Divide" function is also a very usefull tool, so one can break or divide segments that intersect with others. This would be even better if it also removes segments that are independent, without intersections. Check it in QCAD or Allplan.

### Expected behavior

If you pick an independent segment (without intersections) this should be also removed.

### Observed behavior

It only breaks/divides segments where there are intersections.

This is related with #1420

sand1024, 2024-10-24:
hm.... well, actually non-removing such segments was intentional... the action may be changed to support this, yet why? Are you sure it will be more convenient?

gerardbm, 2024-10-24:
Yes. When you have the tool selected and you start breaking lines, sometimes you end up with independent lines that need to be removed as well, so with the same tool you can remove then, without having to use another way.

This feature improves a lot the workflow. Allplan and QCAD have it implemented this way. In AutoCAD is also the same, with the difference that one can keep the mouse pressed and move to the screen to remove any segment hovered in the path of the pointer.

I don't know about other people using it. For me, I've been drawing since 2002 with Allplan using this, and I missed it a long time in AutoCAD. I was happy when QCAD implemented it, and I'm still more happy now in LibreCAD.

With features like this, LibreCAD is seriously a professional tool for CAD. I love it, so thanks a lot for programming this one.

sand1024, 2024-10-24:
oh, I see... well, I'll think about this again, I'm still concerned about mis-clicks, so  probably the most optimal case letting removing them if such entities were selected with CTRL pressed (so at least it's intentional remove). 

> With features like this, LibreCAD is seriously a professional tool for CAD. I love it, so thanks a lot for programming this one.

Well, probably not fully yet - but hope it will. So if you have any suggestions or ideas - please don't hesitate posting them as separate tickets. If they could be implemented - may be one day they will be part of LibreCAD

Questions2000, 2024-10-25:
@sand1024 
Ahhh I had thought that if this was added that it was going to be performed using just left click on items with no intersections and not require the Ctrl modifier.
I understand the mis-clicks concern but not much different then mis-clicking an item to trim/break/divide which causes that piece to delete same as if this request was added.  Can be undone (as long as user is aware they did so :)
When I read this request originally I thought this might have been meant to be because the modifiers were not used yet, I had some crazy ideas for both the Shift and the Ctrl modifiers, but probably would not be added because really they were going to each activate an existing tool.  I believe I have the two posts already typed out/completed for them, I may still post but I will start them both out as closed and maybe just link your name if that is okay.

sand1024, 2024-10-27:
@gerardbm - please check the latest AppImage - in alpha-193 it's possible to delete entities without segment if SHIFT is pressed during click

gerardbm, 2024-10-27:
Hi @sand1024, thanks for implementing this. I'm trying and I'm not able to make it work. BTW, what do you mean with «entities without segment»? Let me show you a video to visualize what I meant.

[https://www.youtube.com/watch?v=-bgJox45wck](https://www.youtube.com/watch?v=-bgJox45wck)

(During the process, after breaking/dividing segments, there are new independent segments that cannot be divided, however they can still be removed with the same function/button. That works with lines, circles, etc. In this video I used QCAD. In Allplan it works the same way).

Instead of the <kbd>Shift</kbd> key, I think it would be better a new checkbox to keep this enabled. So users can active/desactivate it according to their preferences. Or simply leave it by default, because it's really intuitive to keep pressing the button to break or remove segments from the drawing.

ccomin, 2024-11-07:
Hello,

Improvements to the “Divide” tool
Problem:
The “Divide” tool currently allows you to divide a line according to a manually specified dimension. However, it would be useful to add functionality to divide a line into a specific number of sections (e.g. 11), based on its total length, rather than having to enter a dimension.
Suggestion:
Add an option in the “Divide” tool to allow the user to specify the desired number of divisions. The user could then choose between dividing according to a specific dimension or according to the number of divisions along the length of the line.
This would offer greater flexibility and simplify the process for users.

Translated with [DeepL.com](https://www.deepl.com/?utm_campaign=product&utm_source=web_translator&utm_medium=web&utm_content=copy_free_translation) (free version)

sand1024, 2024-11-09:
Well, your considerations are quite reasonable. 
 Please check Slice/Divide (for Line and Circle) commands. https://youtu.be/ILACvThv6I4
 If tick length is set to 0, and divide flag is checked - that command will do just division as you've mentioned.  
 
 Actually, "divide" option is supported by several commands. 
 
Of course, it's possible to have more focused shortcut for these commands in "Modify" section (one that will be concentrate on divide logic only, without drawing ticks). The only thing that this will expand the amount of actions on the screen (in toolbars and left panel), while screen space is limited.

gerardbm, 2025-07-06:
> [@gerardbm](https://github.com/gerardbm) Hm... not sure I got your point there. Could you please make a sample dxf file that reproduces such a situation and attach it (zipped) to this issue? Thus I'll be able to reproduce the problem in my own environment and adjust it.
> 
> Thank you.

Yes, I created a file with an example. In the left side, there's a grid with the remaining points after breaking lines by pressing the shift key (the segments that didn't have any intersection). In the right side, there are no remaining points after this process. The result should be the same of the right side.

[test_break_line.zip](https://github.com/user-attachments/files/21087564/test_break_line.zip)

PS. I had to compress the dxf into a zip file.

sand1024, 2026-09-26:
implemented

closed 2026-09-26 by sand1024