LibreCAD/LibreCAD #1938: Ellipse 1 Point tool incorrect orientation

Questions2000, 2024-10-23:
BUG REPORT

### Expected Behavior
For the ellipse shape shown attached to the mouse cursor to be the shape/orientation that gets drawn when adding it

### Observed Behavior
When you use the Ellipse (1 point) tool it shows the ellipse shaped attached to the mouse cursor with a specific orientation, but when you click mouse to add the ellipse to your drawing it draws it in opposite orientation.

### Steps To Reproduce
-Activate the Ellipse (1 point) tool
-Change its major radius to 3 and its minor radius to 15
-You will see the ellipse shape attached to the mouse cursor appears with a longer vertical length then horizontal length
-Left click to add that ellipse shape to your drawing
-You should notice that when the ellipse is added that it is drawn with the horizontal length longer.
Or in other words rotated 90 degrees from the shape attached to the mouse cursor.

The following works fine (which is the opposite of the above)
-If you change its major radius to 15 and its minor radius to 3
-The ellipse shape attached to mouse cursor is correct
-If you then left click to add that ellipse shape to your drawing
-The ellipse shape added is the correct orientation

### Operating System and LibreCAD Version
LibreCAD-2.2.2_alpha1-181-g8b3cf9ca.AppImage
Linux - Debian 12 based distro
GTK based Desktop Environment

sand1024, 2024-10-26:
fixed in latest alpha 193, please re-test

Questions2000, 2024-10-29:
Tested, this works great, good fix.
I will close this.

closed 2024-10-29 by Questions2000