LibreCAD/LibreCAD #1940: Tangent CC tool requires extra click to apply

Questions2000, 2024-10-23:
BUG REPORT

### Expected Behavior
When using the Tangent CC tool to be able to click on one circle and have the tangent line drawn/applied when the second circle is clicked on.

### Observed Behavior
Not sure if this is intentional.
Currently when using the Tangent CC tool the line does not apply instantly when you click on the second circle, you think that the line is drawn but then realize when you move your mouse, that you need to click on that second circle again to apply the tangent line.

### Steps To Reproduce
-Draw two circles with some space in between (size of circles and space between are not relevant)
-Activate the “Tangent (C,C)” tool
-Click at the top of the circle on the left
-Then move your mouse cursor over to the other circle on the right and you will noticed that a gray line appears to let user know the line that will be created.
-What I expect is that when you click on the second circle (either top or bottom of it) that the tangent line will just be drawn/applied but does not.
-You have to then click on that right circle again to apply the line.

### Operating System and LibreCAD Version
LibreCAD-2.2.2_alpha1-181-g8b3cf9ca.AppImage
Linux - Debian 12 based distro
GTK based Desktop Environment

dxli, 2024-10-23:
@Questions2000 

After selection of circles, if there are more than one solutions, one more step is needed to select one.

sand1024, 2024-10-24:
@Questions2000 
Yes, this is intentional, as there might be several points as @dxli  mentioned. 

However, as far as I remember, I've added similar optimization for such cases (where only single solution exists). Probably one is missing for this action, so I'll take a look. 

So if you see that some tangent (or other) action requires additional click if actually nothing to select (like center of circle, or tangent point) - please report this.

Questions2000, 2024-10-25:
Okay thank you guys
Just want to mention while I am here, the next statement I say might not be what you guys are mentioning so bare with me.

dxli mentioned "if there are more than one solutions"
Meaning user has other options of how the line will be drawn between the two circles.
(which in my example there are other options for where the lines can be drawn, it is not just top of circle to other top of circle)

### Expected Behavior
From my instructions above
`"-Then move your mouse cursor over to the other circle on the right and you will noticed that a gray line appears to let user know the line that will be created."`
I am viewing this as where that decision would be made where to draw the line between the two circles, because the indicators do show up (shows how line will be drawn) at this step in the tool.
Then I am looking at the clicking of the 2nd circle as the confirmation to draw the tangent line.

### Observed Behavior
However when you do click that 2nd circle
It then goes back to what can be viewed as a second showing to the user those line indicators having to make the "decision" of where the line will be drawn again.

To me it seems like what I mention in the "Expected Behavior" is more efficient and easier to understand because user makes the decision that first time those gray guide lines are shown between the two circles.
Rather then having to what seems to be doing the process twice.

Hopefully this makes sense how I am explaining this.

Questions2000, 2024-11-10:
Just wanted to share the following observation as it relates to the issue I mentioned.
I noticed that if you use the `Tangent (P,C)` tool, this works with the expected behavior that I had for the `Tangent (C,C)` tool in this bug report.

### Steps To Reproduce
-Draw circle
-Activate "Tangent (P,C) tool
-Click anywhere to start the line (this would be like clicking the first circle when using the Tangent CC tool)
-Then bring your mouse to the circle you created
-The imaginary lines are shown where line will be created (this is what occurs with Tangent CC tool as well)
-Clicking on that Circle once (which is expected behavior), draws in the current imaginary line that is being shown (this is where the Tangent CC tool requires user to click on that 2nd circle twice to draw the line, which is the bug I was reporting)

Hope this helps