LibreCAD/LibreCAD #1942: LibreCAD closes after Properties dialog size is adjusted

Questions2000, 2024-10-25:
BUG REPORT

### Expected Behavior
To be able to adjust the Properties dialog box size, hit Okay button and have LibreCAD stay open.

### Observed Behavior
LibreCAD instantly closes after Properties dialog box size is adjusted and you hit the Okay button

### Steps To Reproduce
-Draw an object such as a Line
-Double click on that line to open up the Properties dialog
-Adjust the size of that Properties dialog
-Click the Okay button
You will notice that when you click Okay that LibreCAD closes instantly.

### Operating System and LibreCAD Version
LibreCAD-2.2.2_alpha1-188-g6e86fe5d.AppImage
Linux - Debian 12 based distro
GTK based Desktop Environment

app4soft, 2024-10-25:
Duplicate of https://github.com/LibreCAD/LibreCAD/issues/1850
- https://github.com/LibreCAD/LibreCAD/issues/1850

sand1024, 2024-10-25:
please re-test on https://github.com/LibreCAD/LibreCAD/releases/tag/2.2.2_alpha-latest
alpha 190

Questions2000, 2024-10-27:
You got it, tested this out and works great.
Thank you for fixing this as well :)
For some reason it is not letting me tag users
Can someone mention to user app4soft that the post he mentioned above 1850 can also be closed
Thank You

closed 2024-10-27 by Questions2000