LibreCAD/LibreCAD #1944: Statusbar Toolbar - Issues with mouse hover tag for Current Layer

Questions2000, 2024-10-27:
BUG REPORT

### Observed Behavior
In the Statusbar toolbar when you hover your mouse over each item an info tag appears.
I noticed an issue with “Current Layer” info tag.

-If you hover over “Current Layer” text you see a tall narrow empty tag
-If you hover over the <Layer Name> underneath it, then you see a “Name of current active layer" with a very small font text (compared to the font size used in all the other tags).

### Operating System and LibreCAD Version
LibreCAD-2.2.2_alpha1-190-gdfecd0ed.AppImage
Linux - Debian 12 based distro
GTK based Desktop Environment

Hope this helps

IamLRBA, 2024-11-28:
Hello @Questions2000 , is this issue still open?
I'd like to contribute to it

sand1024, 2024-12-18:
@Questions2000 
addressed in last AppImage

Questions2000, 2024-12-20:
Tested and appears to be corrected.
Thank You for fixing this reported issue.

closed 2024-12-20 by Questions2000