LibreCAD/LibreCAD #1945: Circle properties dialog default size is too small

Questions2000, 2024-10-27:
BUG REPORT

If you open the Properties dialog box for a Circle, you will notice that this dialog box is set too small by default, all of its content is condensed in appearance.  The Properties dialog boxes for items like Lines and Polylines are set to open up at a larger size by default, should match them.

Thank You

sand1024, 2024-10-27:
Well, that's know issue with entity dialogs and different operation systems/fonts. While a dialog is fine, say, under Windows, it may have incorrect size under Linux or if other system system fonts are used.  

Therefore, it's possible resize all dialogs. If the setting for dialog persistence is set in Application Preference dialog, if the user adjusts position/size for the dialog  - it will be used on the next dialog's invocation. 

So this is is just annoyance, of course, but actually requires one-time setup (same as the overall customization of the interface).

Probably it will be fixed one day, yet hardly in nearest future.

Questions2000, 2024-10-28:
Oh no problem, I didn't think that setting was related because I remember having it clicked based on a similar issue I had with the Keyboard Shortcuts dialog.  When reading your post above you just reminded me, probably forgot to recheck this option after having to use a new settings file with new AppImage version because of new settings updates.
No need to fix specifically, as long as that setting is there which solves the issue.
I will close, thank you for adding that setting, that is good one :)

Questions2000, 2024-10-29:
@sand1024
(Actually just realized that I cannot test this because there is a bug with Application Preferences which I just reported)

I reopened this post just to confirm something.

Before I mention anything I am fully okay with the solution you mentioned / provided with this setting.
I am just mentioning again just in case because I am not sure I completely follow the other things you have mentioned along with your solution, so just wanted to give a little more information discovered with some more testing.

I tried out some more items Properties dialog boxes

All of the following work fine, they all have normal size dialogs and none of them seem squished
`NORMAL SIZE DIALOGS`
Line
Polyline
Rectangle
Polygon
Spline

`SMALLER SQUISHED DIALOGS`
Circle

(Not as much as circle dialog)
Ellipse
Arc

It appears to be only with curved/round items properties dialogs.
Just trying to understand what in the Circles dialog box would be causing this.
You mentioned Windows OS and fonts sizes but something seems strange.
All the other Properties dialogs consist of the same / similar elements.
I thought it might have something to do with the "Geometry" column, depending on number of fields but that does not seem to be it.  The Line properies have 4 items and the circle has 3 and they are all squished.  Polylines have 1 item and those dialogs are fine as well.
All the ones that work fine all just naturally open up with a larger dialog size.
To me it seems that if one had an issue they all would.
Same as in Windows, would think that circle would be an issue as well (cannot verify)

Again just want to make sure before I close
If you can let me know if I should....
-Close post and use solution mentioned issue is the same
-Keep this issue open now, this new information means it might be something different than expected

Again I am fine with your solution, I apologize but just want to be sure that this issue is not something different.

Thank You

Questions2000, 2025-05-10:
Still occurs but will just use persistent dialog settings.
I will close.

closed 2025-05-10 by Questions2000