LibreCAD/LibreCAD #1946: Garbled text in fonts containing eliptic arcs

SilverGreen93, 2024-10-27:
When opening an AutoCAD exported DXF, some of the text is shown as garbled, the arcs used are not drawn correctly.

### Expected behavior

This is how the text looks in AutoCAD

[image]

### Observed behavior

This is how the text looks in LibreCAD

[image]

### Steps to reproduce or sample file

Use this file to reproduce:
[demo.zip](https://github.com/user-attachments/files/17533282/demo.zip)

### Operating System and LibreCAD version info

Version: 2.2.0.2
Compiler: GNU GCC 13.1.0
Compiled on: Jul 29 2023
Qt Version: 5.12.11
Boost Version: 1.75.0
System: Windows 11 23H2

lordofbikes, 2024-10-27:
based on https://forum.librecad.org/Text-missing-or-garbeled-in-DXF-file-tp5725500.html

dxli, 2024-12-21:
Sorry, pushed to a wrong issue.

https://github.com/LibreCAD/LibreCAD/commit/8f76d523855ce6bf8b931144c704474656f9fb5b

The idea, create all polylines with ellipses to support scaling.

This is clearly experimental, since the change affect all polyline creation.

If this is not acceptable, we will try to limit the changes to fonts only.

dxli, 2024-12-22:
[image]

dxli, 2025-10-31:
Pushed a fix in the 2.2.2 branch

Since the fix is experimental, need to test more before backporting to 2.2.1