LibreCAD/LibreCAD #1948: Orange layer colors are out of rainbow order in the drop down menu

Questions2000, 2024-10-29:
The two newly added orange colors appear to be out of rainbow order in the drop down menu.  They are showing up after magenta.
There are two colors doubled in the location orange would go, which I reported here

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-194-g744fa227.AppImage

Questions2000, 2024-11-29:
Issues
1948 (this one)
1949
1952
are all regarding the adding of an "Orange" layer.
This was kind of left in a strange state, have held off on asking this for a while, but have been wondering....
What does a "restriction" label mean?

sand1024, 2024-12-18:
@Questions2000 
addressed in last AppImage

Questions2000, 2024-12-20:
Tested and appears to be corrected.
Thank You for fixing this reported issue.

closed 2024-12-20 by Questions2000