LibreCAD/LibreCAD #1950: Application Preferences closes LibreCAD with OK button

Questions2000, 2024-10-29:
BUG REPORT
I think this one is important

### Observed Behavior
I noticed that in the Application Preferences if you make any changes and then hit the OK button that it closes LibreCAD completely.
(This appears to be similar behavior to the recently fixed issue with the Properties dialog box closing when hitting OK)

### Steps To Reproduce
For example
-Open up Application Preferences - Defaults
-uncheck `Persistent dialogs location and size` setting
-click on the OK button
you will notice that it crashes LC closes instantly

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-194-g744fa227.AppImage

lordofbikes, 2024-10-29:
duplicate of #1850?

Questions2000, 2024-10-29:
No I do not believe so, as mentioned I think it is similar, but 1850 has been fixed and this issue still occurs/is new.
Coding wise though I am not sure obviously.

sand1024, 2024-11-09:
@Questions2000 
Well, I don't think this is related to previous crashes of dialogs.  
Most probably, it might be related to some specific settings (either changes or just settings values).  
Less probably, it's related specific operating system/window manager. 

In my environment (Win10), I can't reproduce this behavior after lots of different tries. 

So probably you could share settings for librecad somehow (as far as I know, they should be stored in some file under Linux)? 
At least it might be give me some clue of what happens on closing. 

Or, while that might be harder, try to identify which settings values (set or changed) leads to such crash.

Questions2000, 2024-11-10:
@sand1024
Thank you for your reply and for looking into this.
Okay so this does not occur in Windows version, so it must just be Linux version?

-I tried fresh install, no previous settings setup so it does not appear to be linked to specific settings.
- I used the most recent AppImage available `LibreCAD-2.2.2_alpha1-198-g8919ed83.AppImage`

### Steps To Reproduce
-Launch LC AppImage
-Opened “Application Preferences”
-Did not touch any settings (all default settings)
-Just clicked “OK” button
-LC instantly closes

### RUNNING LAUNCH COMMAND FOR LC APPIMAGE IN TERMINAL WITH -d6 RESULTS
Everything is running okay and the last item shown in the terminal is the following
`RS_Math::derationalize: expr = '2.0'`
then when I click "OK" button in “Application Preferences”
LC closes instantly and shows the following two lines at the bottom
```
RS_Math::derationalize: expr = '2.0'
RS_Math::derationalize: expr = '2'
Segmentation fault
```

Not sure if any of this information helps.
If not, would any Linux users be willing to confirm if this happens on your machine or not when using the continuous release AppImage version.
Thank You

@lordofbikes 
You responded above, did you experience this by any chance or was this more of just an observation that the issues could be linked?
Thank You

sand1024, 2024-11-14:
@Questions2000 
Still can't reproduce the issue within my environment, sorry. 
Could you please do the following - make a screenshots for each tab in Application Preference dialog and share them? 
I still suspect that the issue may be related to specific settings, so I hope that will let me identify this case.

dxli, 2024-11-14:
I cannot reproduce this issue with Linux app image.

Might be something local. It helps to try another Linux system, (Like a live CD).

Questions2000, 2024-11-15:
@sand1024
Thank you for continuing to look into this.

@dxli
Thank you for verifying this in Linux AppImage.

At least it is not an issue with the program for everyone, but not good for me because prevents me from being able to set any application preferences.

'LIVE DISTROS/DESKTOP ENVIRONMENTS`
I tested the following AppImages out with a few different Linux distros/DEs with Live versions and same issue occurs with each distro/DE, so it does not appear to be distro/DE related.

`APPIMAGES`
Here some information about the AppImages
LibreCAD-2.2.2_alpha1-190-gdfecd0ed (Last version I have with no close issue, works fine on my machine currently)
LibreCAD-2.2.2_alpha1-191- (I do not have this version)
LibreCAD-2.2.2_alpha1-192- (I do not have this version)
LibreCAD-2.2.2_alpha1-193- (I do not have this version)
LibreCAD-2.2.2_alpha1-194-g744fa227 (First version I have where noticed close issue)

I asked this in the past, does anyone know if there is a location where users can download older versions of the Continuous Releases?
This way maybe I can download 191-193 to narrow down the exact release where this occurs, which may be able to help you guys narrow down if there was something specifically changed in that version that could affect Preferences in this way.

`PREFERENCE SETTING ISSUE ASSUMPTION`
As mentioned above in my other post, I had a fresh computer/LibreCAD setup, no previous config file being used, so all the settings were default, so it might not be specifically setting based?
My workaround is that I can use the 190 version mentioned below, to set my application preferences and then use new version such as current 199 and the new settings are recognized (but still closes LC).  This might prove further that it is not settings based, because newer versions seem to close not matter what settings are set.

`POSSIBLE SOLUTION`
I had a hunch about something when I made the post above, that I might have confirmed.  When I posted this I thought it might have something to do with the ‘2.0’ and the ‘2’

RS_Math::derationalize: expr = '2.0'
RS_Math::derationalize: expr = '2'
Segmentation fault

I ran the `LibreCAD-2.2.2_alpha1-190-gdfecd0ed` (which is the last one that worked) with the "-d6" option and it shows
RS_Math::derationalize: expr = '2'
RS_Math::derationalize: expr = '2'
which maybe confirms that it might have something to do with the ‘2.0’ value?

That is all the information I have at the moment.
Thank you guys again for looking into this.

dxli, 2024-11-15:
I will review the rs_math part.

If that's the cause, it depends on the numerical values, so it explains failures in reproducing the issue.

That rs_math code needs some care anyway.

Does it happen to 2.2.1?

Questions2000, 2024-11-16:
@dxli 
Thank you for looking into this.
To answer your question, no this does not happen when I try this with 2.2.1
`LibreCAD-2.2.1_rc4-3-g9d67596f-x86_64.AppImage`
This release does not have the new updated Application Preferences window.

Also when I run the 2.2.1 AppImage using the "-d6" option it does not mention anything about "RS_Math::derationalize: expr" in the terminal like how the Continuous Release versions do.

Hope this helps
Thanks again

sand1024, 2024-11-16:
@dxli
Well, actually the issue may be not with the dialog itself (and handling OK) -  but somewhere deeper - in applying modified settings after closing the dialog. Really strange behavior... still can't detect what is missing there.

dxli, 2024-11-16:
@sand1024 

I reviewed the code of rs_math::eval(). Couldn't find any crashing issue.

Since this method is essentially used by all GUI methods, the trouble is unlikely limited to app image.

I checked the diff from 91 to 94 as mentioned, but couldn't find anything.

To try snap would be a good idea here, but we do not have snap for 2.2.2

dxli, 2024-11-16:
@sand1024 I will review again only the GUI side, since there's a known good build right before your GUI changes

dxli, 2024-11-17:
@Questions2000 

Currently, our continuous builds are only kept for 2 days. We may consider set it longer.

If the old builds have been deleted, the easiest way to get them is probably by your own fork.

For example, I have my own fork at https://github.com/dxli/LibreCAD
I can add remote at my local repository (need to set up your github ssh keys):

$ git add remote dli git@github.com:dxli/LibreCAD.git

Then, to rebuild the commit 190 dfecd0ed:

$ git checkout -b build190 dfecd0ed
$ git push -f -u dli build190:master

The building will be started, and the results can be downloaded from the Actions tab at you github fork page. The builds are zipped as linuxAssets.zip here.

Could you provide the debug output with the crash? running with "-d6". 

To do that, you need to run the appimage with "--appimage-mount" to mount it first. Something like:

$ /path/to/LibreCAD-*.AppImage --appimage-mount
/tmp/.mount_LibreCklaGnB

Then, 
run the librecad from the mounted folder:
$ /tmp/.mount_LibreCklaGnB/usr/bin/librecad -6

sand1024, 2024-11-19:
@Questions2000 , @dxli 
It seems I've found the reason for this crash, the fix will be in nearest merge request

Questions2000, 2024-11-25:
Glad to that this might have been found.
I am waiting for the next continuous AppImage release to test this out and help you confirm.

sand1024, 2024-11-29:
@Questions2000 - please pick a new AppImage and check whether the issue was fixed.  If not - please let me know, if now the crash is fixed - just please close the ticket. 

Thank you.

Questions2000, 2024-11-29:
@dxli
@sand1024
Thank you guys for taking the time to look into this issue.
I was able to try the new AppImage release yesterday
`LibreCAD-2.2.2_alpha1-207-g7838da61.AppImage`
and this issue now appears to have been corrected.
Thank you for fixing this issue.

closed 2024-11-29 by Questions2000