LibreCAD/LibreCAD #1951: Certain items Properties show ID numbers

Questions2000, 2024-10-29:
### Observed Behavior
I noticed that specific items Properties dialog boxes show ID ### at the top and others do not.
Can someone confirm if this is expected behavior, not sure what these ID numbers mean in this context.

`Some items that show this ID number`
lines
ellipses

`Some items that do not show this ID number`
circles
polylines

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-194-g744fa227.AppImage

Thank You

sand1024, 2024-11-09:
@Questions2000 

Well, those IDs there are internal IDs of entities within drawing. 

They are not too useful right now, yet may be handy in the future. 

Anyway, it's better have consistent way of displaying them. 

Please let me know if you see that ID is missing for properties of entities other than circles and polylines.

Questions2000, 2024-11-10:
Thank you for replying, just letting you know that I recieved your reply and that I went through the properties dialogs and made a quick list, but this lead to some other discoveries about ID numbers so I will hold off on response because trying not to create any new posts for now.

sand1024, 2024-11-11:
@Questions2000 
`so I will hold off on response because trying not to create any new threads for now.`
Hm... why?

Questions2000, 2024-11-12:
I don't know, I just do not want to overwhelm you guys.
I have already taken over the first page of this site.
Just writing to let you know that I have updated the orignal thread post above with some of the items that do not contain ID numbers in Propery dialogs.
I will just make one more post regarding the other issue that I have found regarding ID numbers because as mentioned it kind of relates.
Thank You

sand1024, 2024-11-29:
actually, this is rather a duplicate of LibreCAD#1961

Questions2000, 2024-11-29:
Similar but not duplicate.
1951 = Mentions that some items Properties dialog boxes show ID number at the top and some items Properites dialog boxes 
1961 = Mentions that the Properties dialog boxes that do show ID numbers do not match up with the ID numbers used in the Entity Info widget.

Regardless I am going to close both issues.
Thank You

closed 2024-11-29 by Questions2000