LibreCAD/LibreCAD #1952: Orange layer colors contrast

Questions2000, 2024-10-29:
### Expected Behavior
I tested out the orange layer colors.
It seems that there is very little contrast between the light orange and the dark orange, in comparison to the contrast used with all of the other colors.  If agreed maybe can update with the following, I believe the orange code used is the same, but the dark orange color code is different then the one used.  With the dark orange I used the orange #ffa500 code and brought its fader on the right side of the color picker down to 50% which I believe is how all the other dark color versions for the other colors were achieved.

ORANGE
#ffa500
DARK ORANGE
#7f4e00

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-194-g744fa227.AppImage

Thank You

sand1024, 2024-12-18:
@Questions2000 
addressed in last AppImage

Questions2000, 2024-12-20:
Tested and appears to be corrected.
Thank You for fixing this reported issue.

closed 2024-12-20 by Questions2000