LibreCAD/LibreCAD #1953: Use calculations in custom Dimension text

lordofbikes, 2024-10-30:
### Expected behavior
The measured value of a dimension can be referenced by `<>` in the custom text property.
For detail views, included in a drawing, it might be useful to scale the measured value.
E.g. something like `{{<>*0.5}} mm` which will label the dimension with half of the measured length.

### Observed behavior
This is not possible yet.

### Steps to reproduce or sample file

### Operating System and LibreCAD version info
any

Initiated by: https://forum.librecad.org/Calc-on-Dimension-Label-tp5725520.html
Take a look for more information.

lordofbikes, 2024-10-30:
This sounds compelling, but has some drawbacks to consider.

Implementation can be simple by using muParser which is anyway used all over the UI.

I think the correct and compatible way are VIEWPORTS or PAPERSPACE which are DXF format features and compatible with other CAD software. They should allow to represent portions of the drawing with a different scale, what is the intention of this idea.
Sadly we don't have both in LibreCAD. But their implementation need way more efforts.

Implementing this will make it a LibreCAD only feature. So open files with this feature used, might show dimensions like this `{{60.00*0.5}} mm`, assumed, that `<>` is a common place holder in other CAD too or probably even a (undocumented) DXF standard.
Even older LibreCAD versions will behave so.

Please feel free to share your thoughts, if it's worth the incompatibility.

dxli, 2025-01-02:
{<>*0.5} : use half
{<>*<>}: use squared
{exp(<>)}: use exponential