LibreCAD/LibreCAD #1955: Circle 3 tangents fails to draw if far from the origin

Simusharp, 2024-11-03:
### Expected behavior
I am trying to draw the circle in yellow in the screenshot below. I am using the 3 tangents options to select the circles and the line.

[image]

### Observed behavior
I noticed that if the entities are far from the origin, I am unable to select the third entity.

### Steps to reproduce or sample file

1. The attached file contains two versions of the configuration.
[TangentBug.zip](https://github.com/user-attachments/files/17609142/TangentBug.zip)
2. First work on the copy next to the origin, Use Circle TTT and select the two circles and the line. The circle is drawn correctly.
3. Repeat the previous step with the version far from the origin. The operation fails.

**P.S.:** I believe AutoCAD temporarily transform entities to origin before doing the calculation to avoid problems with entities far from origin (Not sure if this helpful).

### Operating System and LibreCAD version info
Version: 2.2.0
Compiler: GNU GCC 12.2.0
Compiled on: Dec 17 2022
Qt Version: 5.12.11
Boost Version: 1.75.0
System: Windows 10 (10.0)

dxli, 2024-11-05:
@Simusharp 

This is a hard one. Tolerance is always hard in CAD.

For example, if it works around (0,0), moving it away by 100 may still work, but what about 1,000, or even 1,000,000. Eventually, it will break down, due to wasting of significant digits of floating points.

Still there are a few things can be done:

1, this use case requires generic quadratic-quadratic solver, for intersections of two conic sections, and we may continue to apply the pencils of conics to solve a cubic equation instead of a quartic, to improve stability;

2, for the particular use case in the attached dxf. The two conic sections are parabolas with axes of the same direction, and the equation solvers can be reduced to quadratic equation solving, further improve the stability; I can probably have some time to make it work in 2.2.1, depending on my available development hours in the following days.

3, I'm wondering whether the solution could be easier to find, if we explore some projective geometry.

Any suggestion/help will be appreciated.

dxli, 2024-11-05:
@Simusharp 

The problem is intrinsic to the floating point representation.

For example, assume your floating points can only hold 2 digits after the point. The coordinates (1.05, 1.05) is shifted by (10, 10), and the result will be (1.10e1, 1.10e1), shifting it again by (-10, -10), the coordinates will be (1.00, 1.00). Of course, we use double precision, so about 15 digits after the point, but losing of information happens the same way.
The problem is deep with in the equation solver, so small loss of precision in each step causes catastrophic loss of the solutions precision.
My first fix would be switching the algorithm to avoid the direct quartic equation solver. For one line and two circles, a quadratic equation would be much less sensitive to rounding issues.

dxli, 2024-11-06:
The fix is mostly for the case: two circles and one line.

Needed for 2.2.1 also

Simusharp, 2024-11-08:
@dxli, thank you so much for your time. Really appreciate it.

dxli, 2024-11-17:
For future debugging, I added details of this algorithm to the dxf. The second layer(red), shows circle paths, if circles must be tangent to two of the 3 chosen entities.

The circle center path for circles tangent to a line and a circle is a parabola, if the given line and circle do not intersect. Or two parabolas, if they intersects. the path contains two parabolas.

For two given circles, the path is an ellipse, if one circle is within another. If two circles do not overlap, the path is a hyperbola.

[TangentBug.zip](https://github.com/user-attachments/files/17788694/TangentBug.zip)

closed 2024-11-17 by dxli