LibreCAD/LibreCAD #1957: Break Divide tool sets remaining pieces to overridden color

Questions2000, 2024-11-10:
BUG REPORT
I apologize I am trying to hold off on creating new posts but this is one that could cause issues for users using continuous release version.

### Observed Behavior

### Steps To Reproduce

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-198-g8919ed83.AppImage

sand1024, 2024-11-11:
@Questions2000 
`I apologize I am trying to hold off on creating new posts `
Oh, no need to do this - if you see some issue - please report it. 

`drawn items that remain on screen end up having layer color overridden`
Yes, this seems to be a bug... resolved color is set instead of "by layer" one.

sand1024, 2024-12-18:
@Questions2000 
addressed in last AppImage

Questions2000, 2024-12-20:
Tested and appears to be corrected.
Thank You for fixing this reported issue.

closed 2024-12-20 by Questions2000