LibreCAD/LibreCAD #1961: Properties dialogs show different ID numbers than Entity Info widget

Questions2000, 2024-11-12:
BUG REPORT

### Observed Behavior
ID numbers that show up at the top of “Properties” dialogs are different numbers then the ID numbers that are displayed at the top of the “Entity Info” widget.  I think they are probably suppose to be the same?

As mentioned in the following post #1951, some Properties dialogs do not show ID numbers, but the ones that currently do
Line
Ellipse
Arc
Spline
end up showing different ID numbers when compared to Entity Info ID numbers.

### Steps To Reproduce
-Open your “Entity Info” widget tab
-Draw a 2 Point line
-Double click on that line to open its “Properties” dialog
-Compare the number at the top of the “Properites” dialog and the number at the top of the “Entity Info” widget when you hover your mouse over that line

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-199-g8822083c.AppImage

sand1024, 2024-11-16:
@Questions2000 that's not a bug and is a normal behavior (at least, so far), so you may close this issue.  
When properties dialog is closed by "OK", a clone of entity is created and all properties from dialog are applied to that clone. 
And after that, clone replaces original entity and so the ID is changed. 
If properties dialog is closed by "Cancel" - well, the entity should not be changed and IDs should be the same.

dxli, 2024-11-16:
Is it possible to avoid the change, if the dialogue is okayed without any user change?

sand1024, 2024-11-17:
@dxli potentially - yes, but as far as I can see, it will require changes of all dialogs in /ui/dialogs/entity.  There is no common place for setting entity' properties (which is fine, as they are quite different) and there is no way to check whether entity was modified or not. 

So the only thing to achieve this - just modify methods like void QG_DlgArc::updateArc() for each dialog and manually compare value stored in entity's property and value that comes from UI. 

Plus modify bool QG_DialogFactory::requestModifyEntityDialog(RS_Entity* entity) to return false if dialog was closed by OK but entity was not actually modified. 

With such changes, void RS_ActionModifyEntity::trigger() will not create clone and so will not update ID. 

P.S however, in more general sense, I still not sure that the original design with changing entities ID on modifications was a really good idea... as far as I can see, persistent ID's may be more suitable for various features (say, related entities or so). So far I can't see the clear reason why such ID's should be modified, yet probably need to check this deeper.

sand1024, 2024-11-29:
@Questions2000 - settings was added which allows to hide displaying entities' IDs (yet later one ID's may be helpful, so for now just added possibility to hide them from ui as right now ID's are almost useless for the user)

Questions2000, 2024-11-29:
Ahh you read my mind, I noticed this in Application Preferences yesterday and tested it out, works great.
As mentioned I am just going to close both issues 1951 and 1961.
Thank You

closed 2024-11-29 by Questions2000